realsenseai/librealsense #10337: Building a user interface for RealSense depth cameras

Coderooni, 2022-03-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version | Windows 10 |                              | 
| Platform                                 | PC | 
| Segment			| Software Development / RealSense Viewer| 

### Issue Description
This is a generic question and I could probably have gotten an answer if I researched a bit, so apologies if I'm wasting your time. I wanted to program a basic user interface, something similar to the RealSense Viewer, so what would be the best route to take? 

My understanding is since the RealSense APIs are native C++, so that would be the obvious choice to program the user interface/viewer in. Is the RealSense viewer written in C++? I have worked with the RealSense cameras for quite a long time now, using only the python wrappers and only creating console-based applications, so nothing fancy. To sum it up, what language would be the best (in terms of optimal performance) to program a RealSense Viewer clone in? I'm open to suggestions.

MartyG-RealSense, 2022-03-24:
Hi @Coderooni  Don't worry, it is no trouble at all and your questions are welcome.  :)

Yes, the RealSense Viewer is written in C++.

Both C++ and Python are very well supported with example scripts and programming references.  There are a small number of functions that are easier to develop in C++ than Python or easier in Python than C++ but for the application that you are considering developing, you will probably not need to use those functions.  

In particular, if you think that your project will require use of the **software_device()** function then C++ should be the language chosen, whilst if you anticipate needing to put camera stream types on separate pipelines (such as IMU on one pipeline and depth + color on the other) then Python would be an excellent choice.

The RealSense Viewer is open-source that you are free to adapt for your own needs, so if you are less familiar with C++ then using that existing tool as a basis for customization may be easier than writing an application from the beginning.  There are also more C++ programming references for the **Imgui** graphical elements that the Viewer's user interface elements use.

https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer

Coderooni, 2022-03-25:
Hi @MartyG-RealSense, thank you for the swift response.

I'll have a look at the links you've provided. If I were to go with building a GUI in C++, then **Imgui** would be a good place to start, and if I were to build it with Python, what framework would you recommend I choose?

MartyG-RealSense, 2022-03-25:
There is little information regarding programming a GUI in Python, though a couple of RealSense users have previously used **Tkinter** for their GUI interface, as shown at https://github.com/IntelRealSense/librealsense/issues/4657 and the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4406041385875-TKinter-python-with-librealsense2

A tutorial guide for Tkinter is here:

https://realpython.com/python-gui-tkinter/

Alternatively, **Pyglet** may provide a suitable GUI interface, as demonstrated in the RealSense SDK's **pyglet_pointcloud_viewer.py** example program.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/pyglet_pointcloud_viewer.py

Coderooni, 2022-03-25:
Thank you very much Marty, I think these are pretty good starting points for me. If there's anything that needs your attention, I'll check the other relevant issues first before creating a new one.

MartyG-RealSense, 2022-03-31:
Hi @Coderooni  Do you require further assistance with this case, please?  Thanks!

Coderooni, 2022-03-31:
Hi @MartyG-RealSense , nope. I'll close this issue. Thank you.

closed 2022-03-31 by Coderooni