realsenseai/librealsense #10339: How to dismantle D405

pettaa123, 2022-03-25:
Hey folks, I am curious if someone knows how to dismantle D405. The case halfs seem to be glued together? The original case is to big for my use case.

MartyG-RealSense, 2022-03-25:
Hi @pettaa123  On the D435 / D435i and D415 models there are screws on the USB port side-lip that can be undone with an appropriate tool to open the case, whilst on the D455 (which does not have a USB insertion side-lip), it looks as though there are small screws at the bottom of the two mounting holes on the rear of the camera.

Looking down into the two rear mounting holes on the D405 under a strong light, it looks as though there may be little screws in the base of the holes.  However, my understanding is that unlike the circuit boards used in other 400 Series models, the D405's board is not suited to mounting without the outer casing.

MartyG-RealSense, 2022-04-01:
Hi @pettaa123  Do you require further assistance with this case, please?

I had further discussions with Intel about the D401 module.  I am permitted to say that the module will be released for sale separately.  There is not a confirmed release date that I can provide at the time of writing this though.

pettaa123, 2022-04-05:
Here is a video on how we disassemblied it. Of course loses all warranty by doing this. But we needed to get rid of the case. https://youtu.be/dAQVvHwZnsE

Afterwards you need a uncommon 4,5 Torx screwdriver.

closed 2022-04-05 by pettaa123

MartyG-RealSense, 2022-04-05:
Thanks very much for the update and also your guide video and recommendation of an appropriate tool  :)