realsenseai/librealsense #10340: Exception after starting a viewer

divismario, 2022-03-25:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2022-03-26:
Hi @divismario  Please first try downloading and running the 2.49.0 standalone version of the RealSense Viewer - **Intel.RealSense.Viewer.exe** - from the **Assets** file list on the SDK 'Releases' page.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0

[image]

The Intel.RealSense.Viewer.exe file should be run either by left-clicking on the download tab at the bottom of the browser window after download completes or from the folder location on the computer that it was downloaded to (usually the Downloads directory of your user documents folder), and not with the Viewer launch shortcut icon on the Windows desktop.

Once the 2.49.0 version of the RealSense Viewer has been launched, please use the desktop launch shortcut to run the 2.50.0 Viewer and see whether the Access Violation error has disappeared.

MartyG-RealSense, 2022-04-01:
Hi @divismario  Do you require further assistance with this case, please?  Thanks!

divismario, 2022-04-01:
I am able to run 2.49.0 but not 2.50.0 which I'm ok with.

Thanks for your help.
Mario

MartyG-RealSense, 2022-04-01:
Thanks very much @divismario for the update!  As you are okay with using 2.49.0, I will close this case.  Thanks again!

closed 2022-04-01 by MartyG-RealSense