realsenseai/librealsense #10341: Intel-ROS(RDK)

w-1016-Papman, 2022-03-27:
Hi dude, I'm trying to install the Intel-ROS2 project while it's easier to work with my D455 camera and L515 camera, I saw it here[link](https://github.com/ros2/ros2_documentation/blob/rolling/source/Related-Projects/Intel-ROS2-Projects.rst)
And I found this [document](https://intel.github.io/robot_devkit_doc/pages/overview.html)
 , when I followed the steps in 'Installation' step by step to run the command in Ubuntu18.04 (kernel version: Linux4.15) terminal, I found that in my rdk_ws folder there is a missing file called Subfolder of 'turtlebot3_ws'.
I don't know if this is because my system version is too low, I would like to ask if you have understood it, and if you know how to deal with this situation. 
thanks

MartyG-RealSense, 2022-03-27:
Hi @w-1016-Papman  The instructions for that ROS2 robotics project link to this ROS2 wrapper:

https://github.com/intel/ros2_intel_realsense

That wrapper is obsolete though and development on it was stopped before the D455 and L515 camera models were released.  It will therefore not be able to work with your cameras.  It was also created by Intel developers but not by the Intel RealSense team.

The official RealSense ROS2 wrapper that is actively updated is at the link below:

https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta

Regarding why you may have experienced the installation error with the obsolete wrapper, Intel RealSense switched to new secure **https://** web addresses in 2021 instead of using the old amazonaws http addresses.

w-1016-Papman, 2022-03-27:
Your information is very important to me, I will revise it as you said, thank you very much for replying late at night, after all you should be on the other side of the world at this time.

MartyG-RealSense, 2022-03-27:
Don't worry, in my time zone it is early morning now  :)

w-1016-Papman, 2022-03-27:
> 别担心，在我的时区现在是清晨 :)

Hi bro, I'm sorry I might have to disturb you again, but this time I'm already at night.
I'm following the [link](https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta) you gave to install step by step, but when I enter the following command line, a new error appears:
When I enter the following commend:
`rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y`
The terminal error message is as follows：
`WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3`
`
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
realsense2_camera_msgs: Cannot locate rosdep definition for [std_msgs]
realsense2_camera: Cannot locate rosdep definition for [diagnostic_updater]
realsense2_description: Cannot locate rosdep definition for [rclcpp_components]
`
And I found two discussions of related issues, they are here [link1](https://github.com/ros-infrastructure/rosdep/issues/722) , [link2](https://github.com/ros-infrastructure/rosdep/pull/708) . It seems that the warning message doesn't have any effect (But I'm really not sure if the error below is caused by the warning message or has something to do with it), but only from the error message Look, it's because they can't resolve the key
I also encountered some other minor issues, but they were all resolved, but this error message confuses me a bit.
I hope the problem can be solved smoothly here, and of course, I will appreciate any relevant suggestions from you.
thank you!

MartyG-RealSense, 2022-03-27:
As a general rule, if a warning does not have any apparent negative effect on the ROS wrapper's performance (such as being able to smoothly view published RealSense ROS data in the RViz2 interface) then it can likely be safely ignored.  If you have completed the build instructions and you have successfully performed a test launch with **ros2 launch realsense2_camera rs_launch.py** then I would recommend not worrying about the warning.

w-1016-Papman, 2022-03-28:
> 作为一般规则，如果警告对 ROS 包装器的性能没有任何明显的负面影响（例如能够在 RViz2 界面中顺利查看已发布的 RealSense ROS 数据），则可以安全地忽略它。
> 
> 如果您已完成构建说明，并且您已使用**ros2 launch realsense2_camera rs_launch.py**​​ 成功执行了测试启动，并且可以在 RViz2 中查看主题而没有明显减速，那么我建议您不要担心警告。

hi, Dude, I found a similar problem to mine [here](https://github.com/IntelRealSense/realsense-ros/issues/1598) , do you have anything to add?
My system version: Ubuntu18.04.5, and version of ROS2 is eloquent(I am sure that it is supported on my Ubuntu version.)
My librealsense version number is 2.50.0(Build from source and enable it normally after entering realsense-iewer in the terminal)

MartyG-RealSense, 2022-03-28:
As you are using Ubuntu 18.04 with Eloquent, it should be compatible with the current ROS2 wrapper, as shown in the extract below from the wrapper documentation.

[image]

Switching the Python version to Anaconda could cause problems though.  This is because Anaconda installs at a different directory path on the computer to the path used by "ordinary" Python such as Python 3.6.  If librealsense was being used with Conda, usually this would be dealt with by defining a PYTHONPATH to the location of Anaconda on the computer, as mentioned at https://github.com/IntelRealSense/librealsense/issues/10002#issuecomment-981142954  

The warnng **WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3** suggests to me that the ROS wrapper installation cannot find Anaconda and so is setting the Python version as Python 3 instead.

I would recommend not to have the RealSense Viewer opened and streaming data before launching the ROS2 wrapper.  The ROS wrapper does not require the Viewer to be active.

MartyG-RealSense, 2022-04-03:
Hi @w-1016-Papman  Do you require further assistance with this case, please?  Thanks!

w-1016-Papman, 2022-04-03:
Thanks, not needed for now, I think I might open a new issue to discuss if there are other questions

closed 2022-04-03 by w-1016-Papman