realsenseai/librealsense #10342: how can i install librealsense-2.48 by 'sudo apt-get install'???

duxinyu-uestc, 2022-03-28:
i want to know how to install librealsense-2.48 by 'sudo apt-get install'. 
thanks!
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i } | 
| Firmware Version                   |  | 
| Operating System & Version |  ubuntu18.04  | 
| Kernel Version (Linux Only)    |  (4.9)                                         | 
| Platform                                 | nx  |
| SDK Version                            |  {2.48 }                          | 
| Language                            |  {c++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / ubuntu / install / etc..>

MartyG-RealSense, 2022-03-28:
Hi @duxinyu-uestc  If you are using a Jetson Xavier NX, you can use the Jetson package install instructions at the link below to install from packages.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

This will install the latest librealsense version though, which is 2.50.0 at the time of writing this.

If you need 2.48.0 then https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785 suggests a method for installing specific librealsense packages that you could try adapting for the Jetson packages.

It may be better though to install 2.48.0 from source code on your NX though using the RSUSB backend method described for use with Jetson at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

The source code for 2.48.0 can be downloaded from the **Assets** file list at the bottom of the page linked to below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0

[image]

duxinyu-uestc, 2022-03-29:
> Hi @duxinyu-uestc If you are using a Jetson Xavier NX, you can use the Jetson package install instructions at the link below to install from packages.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages
> 
> This will install the latest librealsense version though, which is 2.50.0 at the time of writing this.
> 
> If you need 2.48.0 then [#5773 (comment)](https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785) suggests a method for installing specific librealsense packages that you could try adapting for the Jetson packages.
> 
> It may be better though to install 2.48.0 from source code on your NX though using the **RSUSB backend** method described for use with Jetson at [#6964 (comment)](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049)
> 
> The source code for 2.48.0 can be downloaded from the **Assets** file list at the bottom of the page linked to below.
> 
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0
> 
> [image]

Thank you very much for your help！i can get version 2.48 through your reply.

closed 2022-04-10 by MartyG-RealSense