realsenseai/librealsense #10345: How to make rs-server on Raspberry PI

anthnyprschka, 2022-03-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2022-03-29:
Hi @anthnyprschka  You mentioned at https://github.com/IntelRealSense/librealsense/issues/7123#issuecomment-1040696315 that you tried to open **rtsp://<ip>:8554** in VLC Player.  If you are still using this URL address, are you certain that it is the correct IP address for the stream that you wish to access, please?

anthnyprschka, 2022-03-30:
Hi @MartyG-RealSense, Yes I am certain that I am using the correct IP. In the meantime I was able to open the network device in realsense-viewer. But even on same device I cannot open network stream and VLC namely rtsp://127.0.0.1:8554/RGB Camera/. Any ideas?

MartyG-RealSense, 2022-03-30:
Others have had problems accessing RTSP streams in VLC Player, such as in the case at https://github.com/IntelRealSense/librealsense/issues/7123

It may be that once a stream is being accessed on the RealSense ethernet network via rs-server then an external source such as VLC Player cannot access the same stream whilst it is already being actively used by the Network Device.  The RealSense SDK has a set of rules called **Multi-Streaming Model** that dictate that once a particular stream on a particular camera is 'claimed' then another program cannot access that particular stream until the first program that claimed it releases its claim on it.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

anthnyprschka, 2022-03-31:
Interesting, I also restarted the rs-server and tried to open the stream in VLC only without confirming it worked in realsense-viewer. But frankly the only reason for us to use the rs-server is to allow multiple consumers to consume that RTSP simultaneously which you now say is not possible with rs-server either?

MartyG-RealSense, 2022-04-06:
Hi @anthnyprschka  Do you require further assistance with this case, please?  Thanks!

anthnyprschka, 2022-04-06:
@MartyG-RealSense thanks a lot for the info, we were able to resolve our issue

closed 2022-04-06 by MartyG-RealSense

mhadzic2, 2022-05-26:
Hello @anthnyprschka, I am struggling with the same issue. Do you mind sharing which approach you went with to resolve the issue?