realsenseai/librealsense #10346: D455 not detecting device | Python | Raspberry Pi 3 | Ethernet

Jbilbao12, 2022-03-29:
Hello everyone! After following some of the threads, I don't know if the architecture I have implemented is possible. I have the Framos D455 camera connected to a raspberry pi 3 with the image, raspios-buster-arm64, downloaded from the official website, through the Ethernet port. I ping and get to the camera.

My problem is that the Python Wrapper example does not detect the device.
I have installed librealsense with the following tutorial:
https://github.com/datasith/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

No error, but when I run the pythons I don't detect the camera.

For what reason can it be?

Thanks in advance.

MartyG-RealSense, 2022-03-29:
Hi @Jbilbao12  Intel do not provide support for the FRAMOS 'e' RealSense-compatible industrial camera models as they are not Intel products.  The FRAMOS support site at https://www.framos.com/en/resources/support is the best place to submit your question to.

I would add that the guide that you linked to is designed for Pi 4.  Intel have their own official Pi 3 RealSense installation guide for Raspbian (now called Raspberry Pi OS) at the link below that you could try.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

Intel also have a Pi 3 RealSense guide for the Ubuntu MATE operating system that does not require Python installation.

https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

If using either of the above guides do not resolve your problem (especially as they were designed for Intel-produced RealSense 400 Series cameras) then please contact FRAMOS support for advice.  Thanks!

Jbilbao12, 2022-03-30:
Thank you very much for answering!

I already tried the official installation steps for the Raspberry pi 3 but it is very similar to the 4 and also in the Buster image it gave me errors.
I also put the ubuntu Mate image inside the SD for the raspberry but it was so slow that it was exasperating.

I'll try to write to Framos!

Thank you very much and sorry for the inconvenience!

MartyG-RealSense, 2022-03-30:
It's no trouble at all.  Good luck!

MartyG-RealSense, 2022-03-30:
BTW, Intel provide a pre-built complete ethernet-supporting SD card image for Pi 4 at the link below (from their ethernet networking guide).  My understanding is that it installs the old 2.34.0 version of the RealSense SDK rather than the current 2.50.0 but it may be worth looking at.  The SD card build does not include RealSense wrappers such as Python.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-3-preparing-the-sd-card

Jbilbao12, 2022-03-30:
What should I do with that image? 
In the build folder of librealsense, do " cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=Release"
"make -j1"
The point is that I need the python example.

MartyG-RealSense, 2022-03-30:
I suggested using the SD card image as a means of testing your camera with a librealsense build on Raspberry Pi.  It may not be helpful to you if you have an essential requirement for the Python wrapper to be installed.

In regard to the CMake instruction, including the build flags ** -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3** tells CMake to build the librealsense SDK with the Python wrapper included so that you can run pyrealsense2 scripts.

MartyG-RealSense, 2022-03-30:
You will first need to install the program called **CMake** on your Raspberry Pi.  The link below has a guide for installing it on Pi.

https://lindevs.com/install-cmake-on-raspberry-pi/

MartyG-RealSense, 2022-03-31:
You are very welcome!  Thanks very much for your kind words.