realsenseai/librealsense #10349: Future of Realsense

MartinPedersenpp, 2022-03-30:
Hello
The latest statement I can find regarding the current state of the realsense department and its future is this one:

Hello everyone,
My name is Oded Dubovsky and I am the SW Director of Intel® LibRealSense.
I am posting the following to help alleviate a lot of the stress that the press announcement has caused.

It is true that Intel has decided to wind down the "RealSense" business unit, and is announcing the EOL of LiDAR (L515), Facial Authentication (F450) and Tracking (T265) product lines this month. For these specific products, Intel plans to provide a 6 month EOL and Last Time Buy period for these EOL products. 

However, the Stereo Product Lines WILL continue with the following products: D410, D415, D430, D450 modules and D415, D435, D435i, D455 cameras to our current distribution customers (new distributors will be evaluated on a case by case basis).  Support will be continued for the following Stereo products: D410, 415, 430, 450, modules and D415, 435, 435i integrated product lines. We will also continue the work to support and develop our LibRealSense open source SDK.

In the coming future Intel and the RealSense team will focus our new development on advancing innovative technologies that better support our core businesses and IDM 2.0 strategy.

_Originally posted by @odubovsky in https://github.com/IntelRealSense/librealsense/issues/9653#issuecomment-902682551_

Does anyone know about any official statements or posts that indicate the future of the realsense cameras?
I would like to use the D435(i) for a prototype product, but I don't know if it would be safer to just go with one of the alternatives from the before mentioned post?

Any new info would be much appreciated.

Best regards
Martin

MartyG-RealSense, 2022-03-30:
Hi @MartinPedersenpp  As stated in the above message, the RealSense T265 Tracking Camera, L515 Lidar Depth Camera and the F450 & F455 facial authentication products have been declared end of Life (retired).  The official RealSense Store at the time of writing this is selling L515 and F455 / F450 whilst remaining stocks last.

https://store.intelrealsense.com/
   
The RealSense software tools 'Dimensional Weight Software' (DWS) and 'Touchless Controller Software' (TCS) are also discontinued.

However, the RealSense 400 Series stereo camera range is continuing, and Intel have just added a new **D405** model to the range that specializes in short-range depth sensing (ideal range 7 cm to 50 cm).

MartinPedersenpp, 2022-03-30:
Hello @MartyG-RealSense 

Thank you, I got that from reading the original statement and I am happy to hear you confirm this. I was just a bit insecure because of the "inactivity" on the Realsense website and the fact that I could not google any official information since the quoted post.

I will close this again and hopefully anyone else looking to invest in the D400 series can use this for assurance.

closed 2022-03-30 by MartinPedersenpp