realsenseai/librealsense #10350: rs-server issue with rtsp stream

mgvs, 2022-03-30:
----------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | - | 
| Operating System & Version |   Raspbian  | 
| Kernel Version (Linux Only)    |  4.19.97-v7l+   | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  commit 303403b0525590e4ea54ecdba952f50d26602a0a (HEAD -> master)  17 Feb 2022       | 
| Language                            |  C      | 
| Segment			|  rs-server  | 

Dears, I'm getting issue with RealSense Viewer when I try to add Network Device (raspberry pi + rs D415). I'm able to receive only RGB Camera output, not Stereo Module. Also when I'm trying to get rtsp stream in VLC rtsp://192.168.1.1:8554/RGB Camera/ - I got no output at all.

Thanks

MartyG-RealSense, 2022-03-30:
Hi @mgvs  May I first ask which resolutions and FPS speeds you are setting in the Viewer for depth and RGB?  When the camera is operating as a Network Device it has a more limited range of supported modes, described in the Software Limitations table linked to below.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations

Also, are you using a Raspberry Pi 4 or an earlier model such as Pi 3, please?

mgvs, 2022-03-31:
> Hi @mgvs May I first ask which resolutions and FPS speeds you are setting in the Viewer for depth and RGB? When the camera is operating as a Network Device it has a more limited range of supported modes, described in the Software Limitations table linked to below.
> 
> https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations
> 
> Also, are you using a Raspberry Pi 4 or an earlier model such as Pi 3, please?
> 
> In regard to being unable to access the RTSP stream in VLC Player, another RealSense user reported the same issue today. I suspected at [#10345 (comment)](https://github.com/IntelRealSense/librealsense/issues/10345#issuecomment-1083130744) that it may be due to the Network Device already having a claim on the stream that prevents another program from accessing that stream.

Yes, same issue. Does it solved? The viewer using rtsp://192.168.1.1:8554/RGB Camera/ (got from Wireshark) but if I put it to VLC - doesn't work. Did Intel change anything in rtsp protocol when inherited live555, so VLC unable to work with such streams?