realsenseai/librealsense #10351: Problem with IR stream on D405 - poor quality depth

melsanharawi, 2022-03-30:
I have just bought the Intel D405 a few days ago. Depth quality seems poor on untextured objects because there is no IR projector.
To improve depth accuracy, I use an IR projector (the one provided with intel D415).
The problem is that IR pattern is not visible in infrared stream, and the depth is since not improved (intel viewer v.2.50.0)

[image]
[image]

I attached photos of IR stream from intel D415 then D405.

Why is IR pattern not visible on D405 stream like in D415 IR stream ?
Is there a possibility to improve depth accuracy by IR projection or other projection ?

Thanks

MartyG-RealSense, 2022-03-30:
Hi @melsanharawi  A difference between the D405 and other 400 Series models is that the D405 is equipped with an **IR Cut** filter that blocks infrared frequencies from being visible to the camera.  The D405 will therefore not be able to see infrared from sources such as the projector of a D415.

As 400 Series cameras can alternatively use the ambient light in a scene for depth analysis instead of an IR dot pattern, you could instead try casting a light source onto the surface of the observed object to see whether it improves your depth image.

melsanharawi, 2022-03-30:
Hi @MartyG-RealSense, thanks for your quick reply, I will try to find a light projector with random dot pattern (maybe you can recommend one ?), it should be better, because on textured objects, depth accuracy is good.

MartyG-RealSense, 2022-03-30:
Intel do not have recommendations for specific models of projector, though they have published a guide document for characteristics to seek when choosing a projector for use with 400 Series cameras.  The section of that guide linked to below states that whilst an invisible IR pattern projection is considered ideal, there are scenarios where a visible pattern projection is acceptable.

https://dev.intelrealsense.com/docs/projectors#section-6-the-consumer-front-projector

closed 2022-03-30 by melsanharawi

Yiiipu, 2023-08-31:
why add an IR cut filter? it sounds unnecessary.

MartyG-RealSense, 2023-09-01:
@Yiiipu  There is not a documented reason for the filter's inclusion.

Yiiipu, 2024-02-07:
> Hi @MartyG-RealSense, thanks for your quick reply, I will try to find a light projector with random dot pattern (maybe you can recommend one ?), it should be better, because on textured objects, depth accuracy is good.

Did you find a good visible light projector? I noticed my D405 could see 980nm IR pattern, so I got a 940nm dot projector but have not tried it yet.