realsenseai/librealsense #10352: no device connected : except by using sudo (strange)

PierreLouisT, 2022-03-30:
Hello ! 
I am using a realsense camera D455. 
I am in front of a very strange problem. When I reach my script, it tells me "no device connected" . To pass it I have to use sudo python etc instead of only python and it is problematic because in root mode, I have not library as pyrealsense2 for example. 

I am using ssh to be connected to my computer, and when I don't use ssh connexion, everything is working so I don't know how to solve it, I tried to use 'sudo visudo' but I don't think it is the good way. My main goal is to use service in /etc/systemd/system/ but this problem occurs everytime.. It is not due to cable or port, I think the hardware is well connected, I checked it with lsusb and fdisk

I tried to reinstall Ubuntu environnement and reinstall conda etc, so when I enabled my service, after the reboot, the script stop working and I got "no device connected" again. What are your thoughts about that ? Is it because I use SSH ? the Service ? 

Thank you for your help :)

PierreLouisT, 2022-03-31:
Hi Marty ! I checked it and it didn't change anything, "no device connected" remains over ssh. I need to use sudo python etc.. 
My main problem is that I am using another pc over ssh and my script is running without any problem and I can't understand why because I have done the same installation. I think there should be an administrator problem. Maybe SSH is not well configured

MartyG-RealSense, 2022-03-31:
As there does not appear to be existing librealsense support forum cases about this issue, I took a more general approach of googling for the term **python run script without sudo** to see what factors may prevent a Python script being run without the sudo instruction.

Here is an example link from such a search:

https://www.quora.com/How-do-I-run-a-script-without-Sudo/answer/Eitan-Chmelevsky?ch=10

MartyG-RealSense, 2022-04-06:
Hi @PierreLouisT  Do you require further assistance with this case, please?  Thanks!

PierreLouisT, 2022-04-07:
Hi Marty, sorry for my late, no I found a solution, as expected, it was an admin problem,
Thank you for your help, I close this issue :)

closed 2022-04-07 by PierreLouisT

MartyG-RealSense, 2022-04-07:
No problem at all - I'm pleased that you achieved a solution.  Thanks for the update!

happydrones, 2025-03-05:
Could you share your method for solving this problem? I’ve encountered the same issue, but the difference is that I’m not connecting via SSH to the host. On my Jetson AGX Orin, I only have one user named Oscar, and the default user (Oscar) already has sudo privileges. I’ve been struggling with this for many days.