realsenseai/librealsense #10353: optical Filter for L515

sunwell1994, 2022-03-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.03.00 | 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    |  None                                       | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C }                          | 
| Segment			|  others }                        | 

### Issue Description
There are some discussions on optical filters applied to the D400 series. L515 is affected by the ambient light as well, but there are no instructions on how to select the optical filters for L515.  Can optical filters make L515 more feasible outdoor?  Does anyone have the experience to select an appropriate optical filter? Thnaks

MartyG-RealSense, 2022-03-30:
Hi @sunwell1994  The L515 does not have documentation for use of optical filters and it is also not suited for outdoor use as it operates best in a controlled lighting environment with minimal sunlight.  This is because it is sensitive to infrared light sources.

On a 400 Series camera, the negative impact of glare from reflections can be greatly reduced by applying a thin-film **linear polarization** filter over the camera lenses.  I am not aware of this filter type being tested with L515 lidar depth cameras.  For the 400 Series at least, any polarizer that is linear will be sufficient though so long as it is not a circular polarizer like those used in 3D glasses.  

Basic polarizer film sheets are affordable enough that you could test it with L515 without losing much if the experiment does not work out.  You can search stores such as Amazon for the term **linear polarizing filter sheet** to research example filter products.

Alternatively, you could try setting the L515 **Short Range** camera preset.  This will reduce the values of the Laser Power and Receiver Gain options.  Decreasing laser power can help to avoid over-saturating the receiver, whilst decreasing Receiver Gain can lower noise from ambient light at the expense of maximum range being reduced.

sunwell1994, 2022-03-30:
> **linear polarization** filter

Thanks for your great reply. I have tried *Short Range camera preset*. It will decrease the max range a lot(50cm in my setup. ) 

It is a good idea to try the linear polarization filter. Do you have any suggestions for the passband of  the filter?

MartyG-RealSense, 2022-03-30:
Section **4.4 When to use polarizers and waveplates** of Intel's optical filter white-paper guide discusses linear polarization filters.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-4-the-use-of-optical-filters

Section **3.  Types of optical filters** may provide information about your passband question, as sub-section **3.  Polarizing filter** provides graphical charts of wavelengths for several different types of polarizer (Lingpass, Shortpass, Bandpass and Notch).

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-3-types-of-optical-filters

sunwell1994, 2022-03-30:
> Section **4.4 When to use polarizers and waveplates** of Intel's optical filter white-paper guide discusses linear polarization filters.
> 
> https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-4-the-use-of-optical-filters
> 
> Section **3. Types of optical filters** may provide information about your passband question, as sub-section **3. Polarizing filter** provides graphical charts of wavelengths for several different types of polarizer (Longpass, Shortpass, Bandpass and Notch).
> 
> https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-3-types-of-optical-filters

Helpful! I will take a try.

closed 2022-03-30 by sunwell1994