realsenseai/librealsense #10354: Help with Lidar intensity values on Matlab

zerriet, 2022-03-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   {Win 10 | 
| Platform                                 | PC  |
| SDK Version                            | 2.50.0 }                          | 
| Language                            |  Matlab }                          | 
| Segment			|  LiDAR/Point Clouds/Object Detection }                          | 

Hi there, I am currently trying to test a proof of concept for creating point cloud objects on Matlab by using extracted vertice values from aligned frames using the librealsense Matlab wrapper. The intention is to incorporate recorded test point cloud data into a pre-trained PointPillar detector shown here: [https://www.mathworks.com/help/lidar/ug/object-detection-using-pointpillars-network.html](url).

One issue I currently face is not knowing how to extract intensity values (normalized grayscale intensities at each point) for my point cloud object as I need these to run in tandem with the PointPillar detector. I have tried extracting the infrared frame itself but I not only do not know how to normalize them into a single vector, but I also ended up having more points compared to the number of vertices calculated.  

I would like to get some advice and help on this issue. I would really appreciate it! I have also attached my code with regards to the creation of matlab point cloud objects, as well as accompanying pictures showing the lack of intensity values.

`function [ptCloudArray,timestamps,colordatavector, infrareddatavector] = pcl_dynamic_save(numFrames)

%colorizer object to prettify depth output
colorizer = realsense.colorizer();

% define point cloud object
pcl_obj = realsense.pointcloud();

% Make Pipeline object to manage streaming
pipe = realsense.pipeline();

align_to = realsense.stream.color();

alignedFs = realsense.align(align_to);

% Start streaming on an arbitrary camera with default settings
profile = pipe.start();

% Get streaming device's name

dev = profile.get_device();

name = dev.get_info(realsense.camera_info.name);
serialno = dev.get_info(realsense.camera_info.serial_number);

fprintf(name + "\n" + serialno + "\n");

% Get frames. We discard the first couple to allow
    % the camera time to settle

    ptCloudArray = pointCloud.empty(numFrames,0);
    timestamps = datetime(zeros(numFrames,1), 0, 0); 

    %numframes = numFrames;    
    for i = 1:numFrames
        fs = pipe.wait_for_frames();

        %align depth frames to color stream

        aligned_frames = alignedFs.process(fs);

        depth = aligned_frames.get_depth_frame();
    
        color = fs.get_color_frame();

        infrared = fs.get_infrared_frame();

        pcl_obj.map_to(color);

        infrareddata = infrared.get_data();

        infrareddatavector = [infrareddata(1:3:end)',infrareddata(2:3:end)',infrareddata(3:3:end)'];

        colordata = color.get_data();

        colordatavector = [colordata(1:3:end)',colordata(2:3:end)',colordata(3:3:end)'];

  

         pnts = pcl_obj.calculate(depth);
         vertices = pnts.get_vertices();

         ptCloudArray(i) = pointCloud(vertices);
         timestamps(i) = datetime;

    end

    %stop streaming
    pipe.stop();
    

end

[image]
[image]
 `

MartyG-RealSense, 2022-03-30:
Hi @zerriet   As you are using PCL, it may be worth investigating a PCL class called **RealSense2Grabber** that was designed for use with RealSense 400 Series cameras but might be adaptable for L515.

https://pointclouds.org/documentation/classpcl_1_1_real_sense2_grabber.html

The Grabber documentation lists a function called `pcl:::pointcloud < pcl PointXYZI >` that can convert an Infrared Depth image to a `pcl::PointCloud<pcl::PointXYZI>` (the I in PointXYZI is Intensity).

zerriet, 2022-03-30:
Hi MartyG, thank you for such a quick reply! 

It's awesome that there exists a function that allows for IR to pointcloud-XYZI conversion, but correct me if I am wrong, the one you had linked was referring the use of the pointcloud library (PCL)? I am not sure if I will be able to use it since I am using Matlab as the main platform and not PCL. Hence there would need to exist a similar function found within the Matlab wrapper

MartyG-RealSense, 2022-03-30:
Researching the subject, I see that 'pcl_obj' appears in numerous examples of MATLAB wrapper pointcloud scripts that do not use PCL.  So I apologize for my misinterpretation.
 
Nevertheless, the MATLAB wrapper's scripting language is almost 1:1 with the RealSense SDK's C++ language and so it may be within possibility to interface with PCL from a MATLAB wrapper script.

MartyG-RealSense, 2022-04-08:
Hi @zerriet  Do you require further assistance with this case, please?  Thanks!

zerriet, 2022-04-08:
Hi @MartyG-RealSense! Sorry for the late reply.

I found it a little difficult to integrate PCL onto MATLAB so I was not able to use **RealSense2Grabber** as you had suggested.

Instead, I came up with a workaround solution by calculating pseudo intensity values from the aligned RGB frame. This is based of the INSC RGB to grayscale conversion where the calculation for each intensity value is: 0.299 · Red + 0.587 · Green + 0.114 · Blue.

This was more of a stop-gap solution as it might not be an accurate representation of true reflectance for each data point. Nevertheless, hope it might be useful for people who are currently working using the MATLAB wrapper. I have attached my code for reference. 
` for i = 1:numFrames
        fs = pipe.wait_for_frames();

        %align the depth frames to the color stream
        aligned_frames = alignedFs.process(fs);

        %select aligned depth frame
        depth = aligned_frames.get_depth_frame();   
        %select aligned color frame
        color = aligned_frames.get_color_frame();
       

        
        colordata = color.get_data();

        colordatavector = [colordata(1:3:end)',colordata(2:3:end)',colordata(3:3:end)'];

        grayscalearray = 0.299*colordatavector(:, 1) + 0.587*colordatavector(:, 2) + 0.114*colordatavector (:, 3);                 
                                                                                              
         points = pcl_obj.calculate(depth);
         vertices = points.get_vertices();
         ptCloudArray(i) = pointCloud(vertices,"Color",colordatavector,"Intensity",grayscalearray);
         timestamps(i) = datetime;
    
    end`

Also thank you for all the help provided, this case can be closed.

MartyG-RealSense, 2022-04-08:
Thanks so much @zerriet for sharing your solution with the RealSense community  :)

As you suggest, I will close the case.  Thanks again!

closed 2022-04-08 by MartyG-RealSense