realsenseai/librealsense #10356: How to get rid of gravity in accelerometer data

MyOtherNamesWereTaken, 2022-03-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
Camera Model | { D435i }
-- | --
Firmware Version | (05.13.00.50)
Operating System & Version | { Linux (Ubuntu 14/16/17)
Kernel Version (Linux Only) | (5.11.0 -38-generic )
Platform | PC
Language | {python}
Segment | {Robot/Smartphone/VR/AR/others }

This is my code so far, however it only works (roughly) when the camera is laid flat on the ground with the lenses looking upwards.
I feel like I am lacking some calibration too because my accelerometer measures 9.95 m/s² on average while stationary. I'm wondering if my rotation matrix is correct?
`        
        # gather IMU data
        accel = f[0].as_motion_frame().get_motion_data()
        gyro = f[1].as_motion_frame().get_motion_data()
        ts = f.get_timestamp()

        dt = (ts - last_ts_gyro) / 1000
        last_ts_gyro = ts
        gyro_angle_x = gyro.x  # Roll
        gyro_angle_y = gyro.y  # Pitch
        gyro_angle_z = gyro.z  # Yaw
        # rad/s => rad
        gyro_angle_x *= dt
        gyro_angle_y *= dt
        gyro_angle_z *= dt

        # totals
        total_x += gyro_angle_x
        total_y += gyro_angle_y
        total_z += gyro_angle_z

        # accelerometer calculation
        Qx = np.array([[1, 0, 0], [0, np.cos(total_x), -np.sin(total_x)], [0, np.sin(total_x), np.cos(total_x)]])
        Qy = np.array([[np.cos(total_y), 0, np.sin(total_y)], [0, 1, 0], [-np.sin(total_y), 0, np.cos(total_y)]])
        Qz = np.array([[np.cos(total_z), -np.sin(total_z), 0], [np.sin(total_z), np.cos(total_z), 0], [0, 0, 1]])
        Qxyz = Qz @ Qy @ Qx

        # Q * (a-g) to filter out gravity
        g = np.array([[0], [0], [gravity_norm]])
        accelwithg = np.array([[accel.x], [accel.y], [accel.z]])
        rotated_g = Qxyz @ g
        accelwithoutg = accelwithg - rotated_g`

MartyG-RealSense, 2022-03-31:
Hi @MyOtherNamesWereTaken  The ideal target value for Y acceleration (gravity) is approximately 9.80.  

The IMU component used in RealSense cameras does not have an internal calibration mechanism and so the IMU can be calibrated with a Python software tool instead and have that calibration saved to the camera hardware.  Before an IMU calibration is performed, the Y value may be non-zero when the camera is stationary.

The Python IMU calibration tool can be found at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration

Have you used this tool yet, please?  If you have, the calibration will not be permanently saved unless it is written to the camera hardware at the end of the calibration process.

In regard to your scripting, it is not recommended to manually assign fixed index numbers to gyro and accel as this can cause the values to be reversed (accel values for gyro and gyro values for accel).  Instead, it is recommended that the indexes are assigned **dynamically**, as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/5291#issuecomment-563436161

MyOtherNamesWereTaken, 2022-03-31:
When trying to use that tool this is the error im getting "Error: (25, 'Inappropriate ioctl for device')". What do i do now?

Ill try to look into the comment and adjust my code.

MartyG-RealSense, 2022-03-31:
My understanding is that **Error: (25, 'Inappropriate ioctl for device** indicates that there is a kernel conflict.  Building librealsense with the **RSUSB backend** method that bypasses the kernel may help if you are not using RSUSB already.  

Alternatively, a few RealSense users who recently reported stability problems with 5.11 kernels reported that using kernel **5.11.0-34-generic** resolved their problem.

MyOtherNamesWereTaken, 2022-03-31:
I am using kernel 5.13.0-37-generic at the moment with librealsense 2.50.0

MartyG-RealSense, 2022-03-31:
Thank you.  Is it possible for your project to use 0-34-generic?  If not, using RSUSB backend will enable librealsense to operate without dependance on the kernel whilst not affecting other non-RealSense aspects of your project that might be using that kernel.

If you built librealsense from packages then the build would not be using RSUSB backend.   RSUSB is enabled if building the SDK from source code with CMake and including the build term **-DFORCE_RSUSB_BACKEND=true** in the CMake build instruction.