realsenseai/librealsense #10357: Calibration Tools and API Package for Ubuntu focal

mechatheo, 2022-03-31:
Hello relasense community,

we are trying to obtain the Calibration Tools and API Package for Ubuntu focal (or any version which works actually). We've been following the instructions from your document: https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400_Dyn_Calib_User_Guide.pdf on page 15. 
However either the key is expired or the repos are no longer publicly available. Is there a way to get the package for linux somehow?

Thanks in advance.

MartyG-RealSense, 2022-03-31:
Hi @mechatheo  In March 2021 Intel switched from using **http** addresses to secure **https** addresses and updated the RealSense documentation.  The version of the Dynamic Calibration user guide that you linked to is the July 2020 edition.  There is a newer July 2021 edition of the PDF with the updated https addresses that I have attached to this message.

[Intel RealSense D400 Series Calibration Tools User Guide.pdf](https://github.com/IntelRealSense/librealsense/files/8390301/Intel.RealSense.D400.Series.Calibration.Tools.User.Guide.pdf)

mechatheo, 2022-04-05:
Hello thanks for the quick reply. However i've even tried switching to https, but always end up with the same problem. I did the last attempt using a fresh `ubuntu:bionic` Docker Image and the guide you've posted, i run:

```
apt update && apt install gnupg2 ca-certificates
echo "deb https://librealsense.intel.com/Debian/apt-repo bionic main" > /etc/apt/source.list.d/realsense-public.list
apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
```

and get back `gpg: keyserver receive failed: No name` 

of cause the repo then remains unsigned:

```
Get:5 https://librealsense.intel.com/Debian/apt-repo bionic InRelease [3251 B]
Err:5 https://librealsense.intel.com/Debian/apt-repo bionic InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C8B3A55A6F3EFCDE
```

Was there possibly a key update, one should be aware of?

MartyG-RealSense, 2022-04-05:
The **keys.gnupg.net** keyserver was phased out in 2021.  I see that it is still listed on page 15 of the calibration tool user guide PDF though.  Does the keyserver address below work for you, please?

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key 6F3EFCDE

mechatheo, 2022-04-05:
thank you very much, this resolved the issue

MartyG-RealSense, 2022-04-05:
Great to hear, @mechatheo - thanks very much for the update  :)

closed 2022-04-11 by MartyG-RealSense