realsenseai/librealsense #10359: problem about d435i

flyover-26, 2022-04-01:
hi!
when I run a project I found it need some parameters.
however I don't know how to get there parameters.
It is like

gamma=XXX where XXX is a gamma calibration file, containing a single row with 256 values, mapping [0..255] to the respective irradiance value, i.e. containing the discretized inverse response function. See TUM monoVO dataset for an example.

vignette=XXX where XXX is a monochrome 16bit or 8bit image containing the vignette as pixelwise attenuation factors. See TUM monoVO dataset for an example.

And could you tell me what kind d435i belongs to?because this is important to some parameters selection.

Calibration File for Pre-Rectified Images

Pinhole fx fy cx cy 0
in_width in_height
"crop" / "full" / "none" / "fx fy cx cy 0"
out_width out_height

Calibration File for FOV camera model:

FOV fx fy cx cy omega
in_width in_height
"crop" / "full" / "fx fy cx cy 0"
out_width out_height

Calibration File for Radio-Tangential camera model

RadTan fx fy cx cy k1 k2 r1 r2
in_width in_height
"crop" / "full" / "fx fy cx cy 0"
out_width out_height

Calibration File for Equidistant camera model

EquiDistant fx fy cx cy k1 k2 k3 k4
in_width in_height
"crop" / "full" / "fx fy cx cy 0"
out_width out_height

thank you so much!!!

MartyG-RealSense, 2022-04-01:
Hi @flyover-26  Are you using the **Direct Sparse Odometry** project please, as the information that you provided is apparently quoted from its documentation.

https://github.com/JakobEngel/dso

The **General Notes** for this non-RealSense project warn not to use a camera with a 'rolling' shutter due to the huge geometric distortions that will result.  The D435i camera model has a fast 'global' shutter on its depth sensor but a 'rolling' shutter on its RGB sensor.  The only RealSense camera model that has a fast global shutter on both RGB and depth sensors is the **D455**.

https://github.com/JakobEngel/dso#4-general-notes-for-good-results

In regard to the 16-bit / 8-bit image question, 400 Series cameras produce 16-bit **depth**.  

RealSense **RGB** images are typically 8-bit format, as described in the format list in the SDK's documentation at the link below.

https://intelrealsense.github.io/librealsense/doxygen/rs__sensor_8h.html#ae04b7887ce35d16dbd9d2d295d23aac7

You can obtain the calibration values of your particular camera by launching the rs-enumerate-devices tool in calibration information mode with the command **rs-enumerate-devices -c**

In regard to the gamma calibration file **pcalib.txt**, these seem to be available in zipped files in the **Calibration Sequences** section of the **mono-dataset** at https://vision.in.tum.de/data/datasets/mono-dataset

In the image below, I expanded open the **calib_narrowGamma_vignette.zip** file.

[image]

flyover-26, 2022-04-01:
thank you so much!
  yes ! I want to test DSO with d435i and I only need /color/image_raw which means  only a color monocular camera is needed.
  sorry,I am a fresh man to this.
  In order to get parameter gamma and vignette images,could you tell me what I should do?
  It seems that vignette is a picture in DSO.
  I know a method in https://github.com/tum-vision/mono_dataset_code.
  However I do not know how to apply this method in d435i.
  besides,you mentioned I can get pcalib.txt in the Calibration Sequences.Is this the same to all camera?
 At last,is it ok to use parameters below if I only use monocular color camera?
RadTan fx fy cx cy k1 k2 r1 r2
in_width in_height
"crop" / "full" / "fx fy cx cy 0"
out_width out_height

thank you so much for your help!

MartyG-RealSense, 2022-04-01:
Documentation for this DSO project is here: https://vision.in.tum.de/research/vslam/dso

It states that the main DSO project at https://github.com/JakobEngel/dso is meant to be used with datasets in the [TUM monoVO dataset](https://vision.in.tum.de/data/datasets/mono-dataset) format but not with a live camera.

If you want to use a **live** camera, such as a D435i, then the documentation recommends using a ROS wrapper for DSO at https://github.com/JakobEngel/dso_ros

To access the live data in ROS with this tool though, you need to write your own **Output3DWrapper** using instructions at https://github.com/JakobEngel/dso/blob/master/README.md#34-accessing-data

It looks as though all the information in your original question about calibration files does not have to be done if you are using a live camera instead of a dataset.

https://github.com/JakobEngel/dso_ros acts as a minimal example project for using a live camera, so the expectation is likely to be that you build this GitHub project and then try your camera with it.

flyover-26, 2022-04-08:
thank you so much about your patience.
In project dso_ros,we can see if we want to run a node we need to run command below.

everything as described in the DSO project - only this is for real-time camera input.

	rosrun dso_ros dso_live image:=image_raw \
		calib=XXXXX/camera.txt \
		gamma=XXXXX/pcalib.txt \
		vignette=XXXXX/vignette.png \
So we can see it need gamma and vignette.

MartyG-RealSense, 2022-04-08:
It looks as though after inputting **https://github.com/JakobEngel/dso** in dso_ros, you still need to provide gamma and vignette files from a dataset obtained at https://vision.in.tum.de/mono-dataset

If you go to the **Calibration sequences** file list at https://vision.in.tum.de/data/datasets/mono-dataset you can see that there is a link beside each file called **play (5x)** that plays a small pop-up video that demonstrates the effect applied to the border of the camera image by that particular calibration file.  This should help you to select the appropriate calibration file for your project.  For example:

**narrowgamma_scene1**
[image]

**widegamma_scene1**
[image]

A **vignette.png** image can be obtained from the dataset files under the **Dataset Download** heading of https://vision.in.tum.de/data/datasets/mono-dataset

Again, you can click on the **play (5x)** video links beside the Dataset Download file names to see how your image might look when using the vignette.png file in a particular dataset.

For example, below is the vignette.png image from the **sequence_09** dataset file.

[image]

closed 2022-04-22 by MartyG-RealSense