realsenseai/librealsense #10361: hardcoded UNIX-style file path leads to error under Windows 10

maxrodkin, 2022-04-01:
https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/common/model-views.cpp#L4207
the error in Intel.RealSense.Viewer.exe during connection of D435 cam

`15:27:42 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\model-views.cpp:4225 - Exception caught trying to detect presets: could not open directory: C:\Users\user\Documents\librealsense2/presets/`

MartyG-RealSense, 2022-04-01:
Hi @maxrodkin  Could you first please check whether the camera has its **Advanced Mode** enabled.  This can be performed in the Viewer.  

1.  Left-click on the **More** option near the top of the Viewer's options side-panel to reveal its drop-down menu and look at the **Advanced Mode** menu option. 
 
2.  If it has a tick-mark beside it then this means that Advanced Mode is enabled.  If there is not a tick-mark, left-click on Advanced Mode to place a tick.  Then please test whether the Infrared stream can now be found by the ROS wrapper.

[image]

maxrodkin, 2022-04-02:
hello @MartyG-RealSense , thank you fot your answer but this ,mark has been checked by default. And error stays the same

MartyG-RealSense, 2022-04-02:
Is the **Intel.RealSense.Viewer.exe** Viewer file that you are using one that was downloaded from the **Assets** file list on the RealSense SDK's 'Releases' page, please?

If you are using a Viewer .exe file from the Assets file list and it is the 2.50.0 version, does the error still occur if you download the earlier **2.49.0** version of the file?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0

The Intel.RealSense.Viewer.exe file should be launched from the folder on your computer that you downloaded it ro and not with the launch shortcut icon on the Viewer desktop.

[image]

MartyG-RealSense, 2022-04-16:
Case closed due to no further comments received.

closed 2022-04-16 by MartyG-RealSense

monajalal, 2023-02-13:
Mine is set to advanced mode but still have the problem opening the directory:

[image]
```
(base) mona@ard-gpu-01:~$ uname -a
Linux ard-gpu-01 5.15.0-58-generic #64-Ubuntu SMP Thu Jan 5 11:43:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
(base) mona@ard-gpu-01:~$ lsb_release -a
LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy
(base) mona@ard-gpu-01:~$ ls /home/mona/Documents
total 12K
drwxr-xr-x  4 mona mona 4.0K Jan 18 13:22 Kit
drwxr-xr-x  3 mona mona 4.0K Jan 18 13:22 .
drwxr-x--- 45 mona mona 4.0K Feb  9 16:10 ..
```

monajalal, 2023-02-14:
Thanks @MartyG-RealSense it's really weird how it got fixed after a few times of opening and closing the RealSense Viewer.

MartyG-RealSense, 2023-02-14:
I'm pleased to hear that the problem is fixed for you.  Thanks very much for the update!