realsenseai/librealsense #10363: Using software-device alignment with rs-camera and non-realsense camera

ilyak93, 2022-04-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D465                                                       | 
| Operating System & Version | Win 10                                                    | 
| Platform                                 | PC                                                           |
| SDK Version                           | 2.34.0-0~realsense0.2250                       | 
| Language                               |  C++                                                       | 

Main goal: align realsense camera with nonreal-sense thermal (images mode: RS2_FORMAT_Z16)
The code of the alignment is checked with realsense rgb-depth images and works fine.
I did the calibration of the cameras with matlab:

```
left_images = string(ls("G:\Vista_project\cccalib2\rs_ther_rs_best\"));
left_images = left_images(3:end);
left_images = arrayfun(@(s) append("G:\Vista_project\cccalib2\rs_ther_rs_best\", s), left_images);

right_images = string(ls("G:\Vista_project\cccalib2\rs_ther_thermal_best\"));
right_images = right_images(3:end);
right_images = arrayfun(@(s) append("G:\Vista_project\cccalib2\rs_ther_thermal_best\", s), right_images);

[imagePoints,boardSize] = detectCheckerboardPoints(left_images, right_images);

squareSize = 20;
worldPoints = generateCheckerboardPoints(boardSize,squareSize);

imageSize = [720, 1280];
params = estimateCameraParameters(imagePoints,worldPoints,'ImageSize',imageSize);
showReprojectionErrors(params);
figure;
showExtrinsics(params);
```

and I got a pretty good re-projection error 0f 0.51 and it seems in the intrinsics visualization that the location of the cameras is correct:
 
[image]

The realsense is a little bit above the thermal camera, is in the physical setup, it does seem to be reliable.

Also the reprojection itself  looks good:
[image]

[image]

(the green are the detected points, and they are detected on the rgb image, and on the thermal image its the same locations as in the rgb - i'm not sure if im not plotting them correctly for the thermal images because it looks that it shows me the points of the rgb board on the thermal image, but the reprojection looks perfect)

The resolutions of the cameras are different, as the realsense is 720 x 1280 and the thermal camera is 512 x 640, so I tried two approaches: 

1. Pad the 512 x 640 to 720 x 1280 with zeroes, so the 512 x 640 starts in the left upper corner. 
2. Upsample the 512 x 640 to 720 x 1280 using some simple upsample technique of PIL lib `cv2.resize(tc1_np, size, interpolation=cv2.INTER_AREA)` 

using the code for alignment didn't give correct results.

I also wondered if to use the intrinsics Matlab calculated for realsense or the original from the stream (rs2 api), tried both, didn't get anything significant (although I think the matlab intrinsics were better a little).    

I did hope that the second approach will work, as real-sense seem to be using it, although I don't really know the inner implementation, but all I do know that before the alignment the images are both 720 x 1280, as in this approach. The thing I don't know is how is the calibration parameters are calculated, tho it is the most logical, that they are calculated for both images when they are 720 x 1280 and then `register_extrinsics_to` from the thermal to real-sense should do the job.

So my questions I guess are:
1. How should I deal with different resolution (I think i will also try to lower the rs resolution, so maybe the upsample and downsample will be less harmful for calibration because the initial dimensions will be closer ) ? Do the callibration should be done between the same resolution images, i.e does reals make it that way (make sense cuz then the extrinsics should be the transform from one to another) ?
2. May it be as an indicator of good calibration that Matlab calculated intrinsics should be the same or close to the stream intrinsics ? 
To make sure I'm putting the intrinsics and intrinsics right, here is their structure from matlab:
[image]

Camera1 is realsense and Camera2 is thermal, here are their intrinsics:

[image]
[image]

[... 204 lines cut ...]
The original color image  (I've cut my face on the color image, so there is another upper part)::
[image]

The results (second approach) with rotation as is from the matlab calibration :
[image]

With transposed rotation:
[image]

The second is better, because when I look one image on another, the objects are closer one to another, but it still isn't a good alignment result.

First approach results:
Color:
[image]

Thermal aligned with transposed rotation matrix:
[image]

Thermal aligned with as is rotation matrix:
[image]

Those are at least overlapping (pretty close).

The only thing that comes to my mind I think is to try do better calibration (more images, more positions).
I will be glad for revisiting thing I've done and correct me if I wrong anywhere or any other comments.

MartyG-RealSense, 2022-04-02:
Hi @ilyak93  In answer to your questions:

***************

1.  
I would recommend using the last approach in your case that uses **rs2::align** so that the RealSense SDK's **align processing block** is used for alignment.  The processing block automatically adjusts for differences in resolution and FOV size.

In the case of your change from rs2::align align(RS2_STREAM_COLOR); to **rs2::align align(RS2_STREAM_DEPTH);** it looks as though you are aligning color to depth instead of the much more common depth to color, a method demonstrated by the SDK's C++ **rs-measure** example program for measuring the real-world distance betweeen two points on the image.

https://github.com/IntelRealSense/librealsense/blob/master/examples/measure/rs-measure.cpp#L133

When color to depth alignment is performed, if the color FOV size is smaller than the depth FOV size then the color image will be stretched out to match the depth FOV size.  This can be useful if you want the aligned image to fill the whole screen instead of having a border around it, as typically happens when aligning depth to a color stream with a smaller FOV.

**********

2.  I would say a cautious 'yes' about whether it is a good sign that the intrinsics calculated by MATLAB closely match the stream intrinsics.  I do not know enough about MATLAB calibration to give an absolute answer to this question though.

ilyak93, 2022-04-02:
> I would recommend using the last approach in your case that uses **rs2::align** so that the RealSense SDK's **align processing block** is used for alignment. The processing block automatically adjusts for differences in resolution and FOV size.

Can you detail about that a little bit more ?
I thought that this adjustment is done according too the extrinsics and the intrinsics got by the calibration, so what do you mean by `automatically adjusts`  ?

For now I've noticed that surprisingly the first approach did much better alignment and by moderating a little the ppx and ppy I was managed to `move` the thermal image onto the real sense even more and the alignment seems almost perfect excluding the distortion of the straight lines, small objects and objects around the borders of image. I think that the calibration may be not exact enough although it seems good on the matlab tests and criteria, so that why it was needed a little hand corrections of the ppx and ppy of thermal image as I said. 
I think maybe I can play with the distortion coefficients to achieve same result.

I tried also making undistortion in matlab, where I also did the calibration, so it is the same environment and the arguments of the calibration to the matlab undistort function, i.e:

` `[J,~] = undistortImage(imread(right_images(i)), params.params.CameraParameters2);``

But the result wasn't good because I think as much as the calibration good, it isn't ideal, otherwise i don't know how can I change the result. So maybe playing it by hand and looking at the results on a few dozens images is the best I can do, although it isn't a quantitative approach (like I can't really measure it excluding by eye).

I think that also is a factor using the first approach (excluding the thing you said about automatically adjustment), because stretching 512 x 640 image by up-sampling to 720 * 1280 creates too much noise for the calibration, that's why on the images u can see that the aligned thermal looks like it shrank too much. Of course probably I can then manually adjust the intrinsics (I think maybe the focal length) for controlling this dimensions, but it seems that in both ways the bottleneck is the calibration (or the way that I use this parameters, but I described here the process and that it do looks fine, idk), so that's why the most profitable approach is the one that makes the least changes to the images (no stretching and other pre-proccesing) as in the first approach. From what it seems to me, I guess. 

According to this observation, maybe a good try is the one that I mentioned, to lower the resolution of the realsense to a more similar one to thermal. 

By the way, the same resolution constraint is by using calibration of matlab mainly. Originally the objects of thermal images are just a little bit bigger then the same objects in real sense, that's why I suspect that maybe the first approach gives me better results, because the calibration is easier, thus the parameters are better, and there isn't shrinking effect, just not accurate alignment which I fix by hand (and it do looks good for the big objects in the images).

An exmaple:
[image]

You can notice in the left corner the small device is not exactly aligned, but like the big stuff are aligned, that's already a progress.

MartyG-RealSense, 2022-04-02:
If alignment is performed in your code using rs2::align then the SDK's Align Processing Block mechanism makes automatic adjustments for differences in resolution and FOV size.

If you do not use rs2::align and create your own alignment method then the Align Processing Block is not used and you will have to create your own mechanism for compensating for differences like your various methods described above.

So using rs2::align and having the Align Processing Block automatically handle adjustments between streams is much easier than implementing adjustments yourself.  **But** it is certainly not compulsory to use rs2::align.   What matters in the end is what works best for your project and produces the best results, and if your own align method achieves that then it is totally fine to use it.

closed 2022-04-10 by ilyak93