realsenseai/librealsense #10364: [QUESTION] pyrealsense2 support for python 3.10

devaale, 2022-04-02:
Any ideas if pyrealsense2 will be available for python 3.10 version any time soon?

MartyG-RealSense, 2022-04-02:
Hi @devaale  Whilst support for **pip install pyrealsense2** installation of the pyrealsense2 wrapper with PyPi packages is only available up to Python 3.9 at the time of writing this, you can build the wrapper from **source code** with CMake for newer Python versions such as 3.10.

MartyG-RealSense, 2022-04-16:
Case closed due to no further comments received.

closed 2022-04-16 by MartyG-RealSense

pasandal, 2022-10-21:
Are there any updates on this matter? Is it more than a year since python 3.10 was released. The use of a wrapper is not appropriate in my case.

MartyG-RealSense, 2022-10-21:
Hi @pasandal  Are you asking for pip install support for Python 3.10 because you do not want to install pyrealsense2 from source code, please?

pasandal, 2022-10-21:
Hi, Sorry for my inaccurate wording. Yes, I am requesting pip install support for Python 3.10. In my case, it is not possible to use a **source code** wrapper using Cmake.

MartyG-RealSense, 2022-10-21:
I will enquire to my Intel RealSense colleagues about your question.  Thanks very much for your patience.

There was also a past case at https://github.com/IntelRealSense/librealsense/issues/8229#issuecomment-767115385 where a RealSense user suggested using a **hex editor** tool to edit an existing PyPi **.whl** file to update it for compatibility with newer Python versions.

MartyG-RealSense, 2022-10-24:
Hi @pasandal  My RealSense colleagues responded that Python 3,10 support for PyPi pyrealsense2 installation will be added in a coming release.  I do not currently have a time estimate for implementation.

pasandal, 2022-10-24:
Thanks @MartyG-RealSense . I am happy to here that. Would it be possible to get 3.11 support in the same release? An official version of 3.11 was released today. :rocket:

MartyG-RealSense, 2022-10-24:
I will pass your suggestion on to my colleagues.  Thanks again!

MartyG-RealSense, 2022-10-24:
My colleagues informed me that as Python 3.11 is still in a pre-release stage and may have bugs that need resolving, it will not be integrated yet.  RealSense users can build pyrealsense2 from source code for currently unsupported Python versions such as 3.10 and 3.11, though i appreciate that this is not an option in your particular situation.

tboerstad, 2022-12-01:
@MartyG-RealSense Thank you for keeping us informed. 
When can be expect the coming release with Python 3.10 support?

Python 3.11 was officially released on Oct 24th, adding support for it would make my life easier.

MartyG-RealSense, 2022-12-01:
New releases of the RealSense SDK tend to arrive at durations of 6+ months.  The previous release at the time of writing this (version 2.51.1) was 3 months ago in August 2022.

MartyG-RealSense, 2022-12-06:
Hi @tboerstad  Version **2.53.1** of the Realense SDK is now available and has added Python 3.10 support and pyrealsense2 whl files for 3.10.

**SDK 2.53.1**
https://github.com/IntelRealSense/librealsense/issues/11177

**Python 3.10 PyPi .whl files**
https://pypi.org/project/pyrealsense2/#files

MartyG-RealSense, 2023-01-26:
Hi again @devaale and @tboerstad 

My Intel RealSense colleagues have informed me that Python 3.11 support is on the roadmap for inclusion in a future SDK release.