realsenseai/librealsense #10365: Question: Eye-safety of many D415 cameras used at the same time

puzzlepaint, 2022-04-02:
Hello,

this is probably a dumb question, but since I am unfortunately not knowledgeable about the subject, I would be happy about any comment from the RealSense team on it: Does the use of many D415 cameras at the same time, with their infrared projectors set to maximum intensity, pose any issue for eye-safety?

Checking the [D400 datasheet](https://www.intelrealsense.com/wp-content/uploads/2022/03/Intel-RealSense-D400-Series-Datasheet-March-2022.pdf), it says "The infrared projector meets the Class 1 laser safety standard under normal operation". [Wikipedia states on class 1 lasers](https://en.wikipedia.org/wiki/Laser_safety#Class_1): "A Class 1 laser is safe under all conditions of normal use. This means the maximum permissible exposure (MPE) cannot be exceeded when viewing a laser with the naked eye or with the aid of typical magnifying optics (e.g. telescope or microscope)."

However, I haven't found a statement on using multiple devices at the same time. For example, using 32 D415 devices in a setup for a volumetric video capture stage, all pointing towards the center of the stage. As a layman, I would guess that the safety of class 1 lasers viewed with "typical magnifying optics" means that at the very least the use of multiple class 1 lasers without any magnifying optics should definitely be safe. But I would be happy about a comment from someone who is knowledgeable rather than relying on my layman's guessing.

Thanks!

MartyG-RealSense, 2022-04-03:
Hi @puzzlepaint  The number of 400 Series cameras with projectors equipped that are present in a volumetric setup should not make a difference to eye safety.  

If you are still cautious though, if the scene is well lit then you may be able to disable the projector's laser completely.  This is because 400 Series cameras can alternatively use ambient light in the scene to analyse objects / surfaces for depth information instead of using the projector's dot pattern projection.

If you feel that the projector will be required, Intel's white-paper document about projector use with 400 Series cameras states that increasing the Laser Power of the camera's in-built projector from its default of '150' to its maximum of '360' only provides a limited improvement in range.  You could therefore likely achieve satisfactory results with the default Laser Power setting without maximizing it.

https://dev.intelrealsense.com/docs/projectors#section-4-increasing-range

Bear in mind that no matter how many cameras you have in the scene, there will only be one camera at a particular position in the room and so only one laser source from that particular trajectory, not multiple overlaid emissions on the same path.

Point number **3.  Use multiple projectors** of the Intel projector white-paper section linked to above confirms that "if each individual projector is rated eye-safe, more can actually be added safely, as they do not originate from the same point source".

***********

You may be interested in the large 100 camera volumetric space called Intel Studios that Intel created at LAX airport in the United States.  Whilst the cameras used are not RealSense and the scale of the room may be larger than your planned volumetric project, the YouTube video at the link below may provide some useful insights about implementing a setup with tens of cameras.

https://www.youtube.com/watch?v=76sehizdvjI

puzzlepaint, 2022-04-03:
> Bear in mind that no matter how many cameras you have in the scene, there will only be one camera at a particular position in the room and so only one laser source from that particular trajectory, [...]

That makes a lot of sense and clarifies my question. Thank you very much for the detailed reply (and also thanks for the additional info about the Intel Studios setup)! Issue resolved.

closed 2022-04-03 by puzzlepaint

MartyG-RealSense, 2022-04-03:
You are very welcome  :)