realsenseai/librealsense #10370: Is building on kernel 5.8 possible?

CreaM129129, 2022-04-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | | 
| Operating System & Version |  Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  (5.8.18-050818-generic)                                         | 
| Platform                                 | PC |
| SDK Version                            |  SDK 2.0 (v2.50.0) }                          | 

### Issue Description
Hello.
From [sdk 2.50](https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0) and some related issues, I have understood, that installing via packages is possible on 5.8 kernel on ubuntu 20.04.

However on [linux instalation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) it is written 5.8 is not supported. Can you update this if it is the case?

While folowing instructions from the link above there are some errors during the instalation:

```
sudo apt install librealsense2-dkms

Reading package lists... Done
Building dependency tree       
Reading state information... Done
librealsense2-dkms is already the newest version (1.3.18-0ubuntu1).
Skipping librealsense2-utils, it is not installed and only upgrades are requested.
0 upgraded, 0 newly installed, 0 to remove and 51 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up librealsense2-dkms (1.3.18-0ubuntu1) ...
Removing old librealsense2-dkms-1.3.18 DKMS files...

------------------------------
Deleting module version: 1.3.18
completely from the DKMS tree.
------------------------------
Done.
Loading new librealsense2-dkms-1.3.18 DKMS files...
Building for 5.8.18-050818-generic
Building initial module for 5.8.18-050818-generic
ERROR (dkms apport): kernel package linux-headers-5.8.18-050818-generic is not supported
Error! Bad return status for module build on kernel: 5.8.18-050818-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.18/build/make.log for more information.
dpkg: error processing package librealsense2-dkms (--configure):
 installed librealsense2-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 librealsense2-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Does this mean kernel 5.8 is truly not supported? 
I assume the instalation guide installs latest version of SDK. Is this true? If not how can I specify I want to install the latest version?

Thanks in advance!

MartyG-RealSense, 2022-04-06:
Hi @CreaM129129  Yes, the official installation guides will install the latest version of the librealsense SDK.

https://github.com/IntelRealSense/librealsense/pull/8787 states that SDK 2.45.0 added support for kernel 5.8 on Ubuntu 20.04 (Focal) but it did not state whether that only applied to patching a source-code build or DKMS packages too.  Currently at the time of writing this, the package distribution installation instructions only list official support up to kernel 5.4 and so I recommend taking that as fact.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

[image]

If you wish to use kernel 5.8, I would therefore recommend building the SDK from source code if possible.  You can do this using the **manual installation** guide at the link below, which you can use to build the SDK with the **V4L (native) backend** after you have applied a patch to the 5.8 kernel.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Alternatively, you could build from source code with the **RSUSB backend**.  This mode of building the SDK is not dependent on Linux versions or kernel versions and does not require patching, so it should not matter which kernel version you are using as an RSUSB build bypasses the kernel.

CreaM129129, 2022-04-06:
I see. Thanks for clearing this out. Closing the issue.

closed 2022-04-06 by CreaM129129