realsenseai/librealsense #10371: Gyro values keeps moving?

MyOtherNamesWereTaken, 2022-04-06:
Camera Model | { D435i }
Firmware Version | (05.13.00.50)
Operating System & Version | { Linux (Ubuntu 14/16/17)
Kernel Version (Linux Only) | (5.11.0 -38-generic )
Platform | PC
Language | {python}
Segment | {Robot/Smartphone/VR/AR/others }

For some reason my Gyro value on Y keeps going up while X & Z go down. Camera lays on a flat & steady surface with the Z axis pointing upwards.
try:
    last_frame = None
    while True:
        f = pipeline.wait_for_frames()
        for frame in f:
            # gather IMU data
            if frame.is_motion_frame():
                if frame.get_profile().stream_type() == rs.stream.gyro:
                    gyro = np.array([gyro_data(frame.as_motion_frame().get_motion_data())]).T
                if frame.get_profile().stream_type() == rs.stream.accel:
                    accelwithg = np.array([accel_data(frame.as_motion_frame().get_motion_data())]).T

                ts = f.get_timestamp()

       if first:
            first = False
            last_ts_gyro = ts
            continue

        if ts == last_ts_gyro:
            continue

        ## calculation for the second frame onwards
        # gyro-meter calculations
        dt = (ts - last_ts_gyro) / 1000
        last_ts_gyro = ts

        gyro_angle_x = gyro[0] * dt
        gyro_angle_y = gyro[1] * dt
        gyro_angle_z = gyro[2] * dt
        # totals
        total_x += gyro_angle_x
        total_y += gyro_angle_y
        total_z += gyro_angle_z

        # rad = > degree
        dtotal_x = (total_x * 57.2958) % 360
        dtotal_y = (total_y * 57.2958) % 360
        dtotal_z = (total_z * 57.2958) % 360

MartyG-RealSense, 2022-04-06:
Hi @MyOtherNamesWereTaken  Have you performed a calibration of the IMU using Intel's Python calibration tool at the link below and then saved that calibration to storage inside the camera, please?  

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration

The IMU component inside 400 Series cameras equipped with an IMU does not have an internal calibration mechanism and so calibration is performed with software instead and saved to the camera once you are satisfied with the calibration result.

If an IMU has not had a calibration written then the Y accel value may be non-zero even when the camera is stationary.  It may also have difficulty approaching the ideal Y value of 9.80 (representing the 9.81 Newtons gravity force of the Earth) and have less accurate values such as 9.4 or 9.6 instead.

MyOtherNamesWereTaken, 2022-04-06:
waiting for realsense device...
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  902512070280
  Product Line:  D400
  Firmware version:  05.13.00.50
Start interactive mode:

Error: (25, 'Inappropriate ioctl for device')

Process finished with exit code 0

this is what im getting when i try to run it

MartyG-RealSense, 2022-04-06:
That is the problem that you reported a week ago at https://github.com/IntelRealSense/librealsense/issues/10356 when asking about accel.  Did you try an RSUSB build of librealsense as suggested in that case, please?

MyOtherNamesWereTaken, 2022-04-06:
How would i go on about doing that? I am a little clueless with this

MartyG-RealSense, 2022-04-06:
When you built librealsense the last time, did you build it from source code using CMake and build the Python wrapper at the same time by including the CMake build term **-DBUILD_PYTHON_BINDINGS:bool=true**
  
If so then you can build librealsense in RSUSB mode by adding the term **-DFORCE_RSUSB_BACKEND=true** to the CMake build instruction.  For example:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS:bool=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

When building with RSUSB, you do not need to apply a patch to the kernel before you build librealsense.

In regard to checking if you were previously using RSUSB: if you built librealsense from source code on a PC and did not use  -DFORCE_RSUSB_BACKEND=true then the build was not using RSUSB.

MyOtherNamesWereTaken, 2022-04-06:
I just used import pyrealsense2 as rs?

MartyG-RealSense, 2022-04-06:
On Linux the Python pyrealsense2 wrapper has to be built by some method (pip install pyrealsense2, building it from source code on its own, or building librealsense and the wrapper together from source code at the same time).  A pyrealsense2 script should not work if the wrapper has not been built.

For example, if librealsense was built from packages instead of source code then it should not have pyrealsense2 included.

Can you describe the method that you use to install librealsense, please?

MyOtherNamesWereTaken, 2022-04-06:
It was build by someone prior to me on Linux with the pip install pyrealsense2 is what ive been told if that helps..?

MartyG-RealSense, 2022-04-06:
That is helpful, yes.

The easiest way to perform a librealsense build that has pyrealsense2 support may therefore be:

1.  Build the SDK without pyrealsense2 included using the simple **libuvc backend** method in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

2.  Once the SDK as been successfully installed, perform the **pip install pyrealsense2** procedure to add the pyrealsense2 wrapper.

MyOtherNamesWereTaken, 2022-04-07:
So i did step 1-5 from the link u send me and then did the pip install but it still gives me the same error.

MartyG-RealSense, 2022-04-07:
Okay, thanks for testing.  So the librealsense SDK is installing correctly but the problem is occurring when installing the pyrealsense2 wrapper afterwards?

MyOtherNamesWereTaken, 2022-04-07:
As in after trying to run the calibration tool im still getting the same error as i did before. Did i do something wrong in the process? Sorry for being this clueless.

MartyG-RealSense, 2022-04-07:
Don't worry, it's no trouble at all.

If you installed librealsense using the libuvc_installation.md instructions then your librealsense build should now be based on RSUSB.  However, you are still experiencing the **Inappropriate ioctl for device** error.  This would suggest that the problem is not related to the kernel, as RSUSB should be bypassing the kernel.

I have not used this calibration tool myself.  My interpretation of the instructions for running the calibration script is as follows though:

1.  Download the source code of the librealsense SDK.  This can be done from the 'Assets' file list of the SDK **releases** page by clicking on the **Source code (zip)** link.

[image]

2.  Once the source code zip file is downloaded, extract the librealsense folder from it and navigate to the following folder location:

**librealsense-2.50.0 > tools\= > rs-imu-calibration**

In this folder should be the script file **rs-imu-calibration.py**

3.  Whilst in this folder, input the Python command below to launch the script.

**python rs-imu-calibration.py**

********

Is the above procedure similar to what you have been doing, please?

MyOtherNamesWereTaken, 2022-04-07:
Thank you so much! It actually opens and does stuff now. Now I just go by the readme file included in the rs-imu-calibration folder, right?

MartyG-RealSense, 2022-04-07:
Great news that you made significant progress!

Yes, you can use that readme.  In my opiinon though, the PDF version of the guide with more detailed instructions provides a clearer explanation of each step of the calibration process.  You can find it at page 16 onwards of the PDF.

https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

MyOtherNamesWereTaken, 2022-04-07:
Thank you so much. I managed to calibrate it thanks to you. Am i supposed to save the calibration files in a certain spot ? Or just the folder the .py is in?

MyOtherNamesWereTaken, 2022-04-11:
So i did save it in the camera folder, however my acceleration is still measured at 9.41 and when comparing the results to those without calibration it seems like nothing has changed. Am i supposed to move the build folder into my projects folder where my python code is?

MartyG-RealSense, 2022-04-11:
I believe that you should use the instruction **RS2_OPTION_ENABLE_MOTION_CORRECTION** to enable motion data correction.

The RealSense SDK's **rs_imu_calibration.py** script provides an example of setting the status of this option in Python, though you should set the option to '1' (enabled) instead of '0' (disabled).

https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-imu-calibration/rs-imu-calibration.py#L264-L266

MyOtherNamesWereTaken, 2022-04-11:
That fixed the issue, thanks! Do you happen to know how accurate the gyro & accelerometer are on this device?

MartyG-RealSense, 2022-04-11:
When **rs_imu_calibration.py** asks if you want to store the IMU calibration at the end of the calibration process and you select Yes, it is saved to an internal storage space inside the camera and not to the computer.  If the calibration is not saved then it will be lost when the calibration program is closed or you perform another calibration.

MyOtherNamesWereTaken, 2022-04-11:
> That fixed the issue, thanks! Do you happen to know how accurate the gyro & accelerometer are on this device?
> 
> Nevermind, it seems that when putting it on different axis it measures different values for the acceleration, Varrying between 9.4 and 10.1 m/s²

I just recalibrated it, however im still facing these issues. I also enable motion correction, which did help.

MartyG-RealSense, 2022-04-11:
IMU data can be noisy and sensitive.  It is possible to write code to weight the data towards the gyro instead of the accelerometer to make it less sensitive.  There is only a C++ example of this though, in the SDK's **rs-motion** IMU program.

https://github.com/IntelRealSense/librealsense/blob/master/examples/motion/rs-motion.cpp#L117-L118

MyOtherNamesWereTaken, 2022-04-11:
Also, when using it like this 

context = rs.context()
pipeline = rs.pipeline(context)
config = rs.config()
rs.option.enable_motion_correction, 1
config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)
config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 63)
profile = pipeline.start(config)

My IDE says that the motion correction statement seems to have no effect?

MartyG-RealSense, 2022-04-11:
The script does not seem to be informing the camera that it should be accessing the IMU component of the camera using **imu_sensor.set_option** to set the motion correction, as the IMU component is separate from the depth sensor inside the camera.  The script therefore needs to define which of the three separate sensor components in the D435i (depth, RGB, IMU) to access.

`self.imu_sensor.set_option(rs.option.enable_motion_correction, 1)`

MartyG-RealSense, 2022-04-11:
Thanks very much for your patience while I conducted further research.  It seems that RS2_OPTION_ENABLE_MOTION_CORRECTION is a **software** setting that tells the script whether or not to correct the motion data and not an option that needs to access the IMU sensor.   So your original implementation **rs.option.enable_motion_correction, 1** was likely correct.  I do apologize.

MartyG-RealSense, 2022-04-11:
https://github.com/IntelRealSense/realsense-ros/issues/2268#issuecomment-1058182675 describes how **VINS-Fusion**, which some RealSense ROS D435i owners have used, has a parameter called **g_norm** to adjust the gravity magnitude value.

I understand that the **robot_localization** ROS module used in Intel's D435i SLAM guide can also provide gravity removal.  An example reference for this is at https://github.com/cra-ros-pkg/robot_localization/issues/501

MyOtherNamesWereTaken, 2022-04-11:
So theres no Python implementation to get a proper rotation matrix? or any sort of method to filter out the gravity?

MyOtherNamesWereTaken, 2022-04-11:
So: in other words theres no way to reliably track position just with gyro and accelerometer?

MartyG-RealSense, 2022-04-11:
Even with a specialized T265 Tracking Camera (see the link below) it is unlikely that you would achieve that very fine degree of position tracking accuracy, so for a D435i I would say no.

https://support.intelrealsense.com/hc/en-us/community/posts/360036372154-T265-Tracking-Accuracy

MyOtherNamesWereTaken, 2022-04-13:
Thanks, ill have to find a different method to make this work. 

Is there any sort of implementation in the realsense library for visual odometry? I couldn't find anything related to Python

MartyG-RealSense, 2022-04-13:
There is support for visual odometry in the RealSense SDK but it is for the T265.

https://www.intelrealsense.com/visual-inertial-tracking-case-study/

A T265 can be paired with a 400 Series camera to provide Visual SLAM navigation by the T265 whilst the 400 Series camera provides depth sensing, as the T265 does not have built-in depth sensing.

https://www.intelrealsense.com/depth-and-tracking-combined-get-started/

MyOtherNamesWereTaken, 2022-04-13:
So the only way to achieve visual odometry with my d435i is with pairing it to a T265?

MartyG-RealSense, 2022-04-13:
The commercial software package **SLAMcore** is compatible with D435i and D455.

https://www.slamcore.com/spatial-intelligence-sdk
https://www.youtube.com/watch?v=TqgwCGrqGAM

A RealSense user performed visual odometry SLAM with a D435i and **RTABMAP** in the YouTube video below.  Others have used RTABMAP with D435i for visual odometry too.

https://www.youtube.com/watch?v=0M_Hc0pdgcY

My understanding is that T265 is the preferred method though due to the noisiness of the D435i IMU data.

MartyG-RealSense, 2022-04-19:
Hi @MyOtherNamesWereTaken  Do you require further assistance with this case, please?  Thanks!

closed 2022-04-19 by MyOtherNamesWereTaken

JACKLiuDay, 2022-06-13:
> Hi @MyOtherNamesWereTaken Do you require further assistance with this case, please? Thanks!

Hi, guy. I need your help. When I run the rs-imu-calibration, it asks me : Would you like to write the results to the camera?(Y/N)
I chose y. But it shows : Writing calibration to device.
Then: Error : failed to set power state.