realsenseai/librealsense #10372: Sometimes SEGFAULT because of NULL pointer dereference in set_power_state with RSUSB backend

mlederhi, 2022-04-06:
Camera Model: D455
Firmware Version: 05.13.00.50
Operating System & Version: Gentoo Linux running Kernel 5.10.61
Platform: PC
SDK Version: librealsense-2.50.0
Language: C++

### Issue Description
When I detect that the connection to the camera was lost (because I unplugged it) then I destruct the old rs2:pipeline object, and construct a new one. Sometimes I get a SEGFAULT because of a NULL pointer dereference in backend.h:696 because 'elem' is a nullptr.

```
void set_power_state(power_state state) override
{
    for (auto& elem : _dev)
    {
        elem->set_power_state(state);
    }
}
```

The pull request #9324 from somebody else fixes the problem for me. Please review it and fix the bug.

Thank you,

Martin Lederhilger

MartyG-RealSense, 2022-04-06:
Hi @mlederhi  The best way for RealSense users to implement the fix suggested by the pull https://github.com/IntelRealSense/librealsense/pull/9324/commits/e38d83df6d11f7e497316ab3682d9c72760bed09 in their own projects will be for them to implement it themselves as you did rather than waiting for a pull merge..

MartyG-RealSense, 2022-04-06:
I would add that the librealsense SDK does not support kernel 5.10 (it can support 5.8 and 5.11).  The SDK can work with non-supported kernels but there may be unprdictable consequences in regard to stability unless the SDK is built from source code with the RSUSB mehod to bypass the kernel.

mlederhi, 2022-04-07:
Thank you for your answer.

I have corrected it myself in the meantime, but of course it would be good that the upstream project also gets fixed so that others can benefit (and me too when the next version is released).

But this problem has nothing to do with the operating system. This is clearly a bug in librealsense. Silently returning nullptrs in case of an error is a no go as this leads to a SEGFAULT later on.

MartyG-RealSense, 2022-04-07:
This support forum is not involved in the assigning of reviewers to pull requests, so I cannot provide a time estimate for when a reviewer may be assigned or a merging into the SDK completed once a pull is reviewed.

MartyG-RealSense, 2022-04-14:
Hi @mlederhi  Bearing in mind the information provided in the comment above, do you require further assistance with this case, please?  Thanks!

mlederhi, 2022-04-14:
No, thank you.

MartyG-RealSense, 2022-04-14:
Ok, thanks very much for your understanding.  I will close this case.  Thanks again!

closed 2022-04-14 by MartyG-RealSense