realsenseai/librealsense #10373: Error while installing python during make -j4 command.

VanKush, 2022-04-07:
[ 60%] Building CXX object src/gl/CMakeFiles/realsense2-gl.dir/__/__/third-party/easyloggingpp/src/easylogging++.cc.o
CMakeFiles/Makefile2:819: recipe for target 'src/gl/CMakeFiles/realsense2-gl.dir/all' failed
make[1]: *** [src/gl/CMakeFiles/realsense2-gl.dir/all] Error 2
[ 60%] Built target pyrealsense2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Can you please help me ??

MartyG-RealSense, 2022-04-07:
Hi @VanKush   Which computer or computing device are you using, please (Raspberry Pi, Nvidia Jetson, Mac, PC desktop / laptop, etc,).  Thanks!

VanKush, 2022-04-07:
I am using PC Desktop with Ubuntu 16.04. along with anaconda version- 4.11.0 an environment  with python 3.7.

VanKush, 2022-04-07:
[image]

VanKush, 2022-04-07:
I am following the instruction given on this link :
https://github.com/IntelRealSense/librealsense/issues/7722

MartyG-RealSense, 2022-04-07:
The linked-to instructions are for an **Nvidia Jetson** caseless computer board, which have a different processor chip architecture (Arm) to that of PC computers (which typically have an x86 / x64 architecture processor) and so Jetson instructions are not recommended for PCs.

As you have Ubuntu 16.04, the simple **libuvc backend** installation method for the librealsense SDK may suit your needs if you have not successfully installed librealsense already.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

You could then build the pyrealsense2 Python wrapper afterwards using the simple command **pip install pyrealsense2** to install the wrapper from a PyPi package instead of source code.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

VanKush, 2022-04-07:
I have tried the installation process with you give link but still facing the similar kind of error.

[ 57%] Building CXX object src/gl/CMakeFiles/realsense2-gl.dir/align-gl.cpp.o
In file included from /usr/include/c++/5/backward/strstream:50:0,
                 from /home/irm2016002/librealsense_build/librealsense-master/src/gl/align-gl.cpp:23:
/usr/include/c++/5/backward/backward_warning.h:32:2: warning: #warning This file includes at least one deprecated or antiquated header which may be removed without further notice at a future date. Please use a non-deprecated interface with equivalent functionality instead. For a listing of replacement headers and interfaces, consult the file backward_warning.h. To disable this warning use -Wno-deprecated. [-Wcpp]
 #warning \
  ^
[ 57%] Building C object src/gl/CMakeFiles/realsense2-gl.dir/__/__/third-party/realsense-file/lz4/lz4.c.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libGL.so', needed by 'src/gl/librealsense2-gl.so.2.50.0'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 57%] Building CXX object src/gl/CMakeFiles/realsense2-gl.dir/__/__/third-party/easyloggingpp/src/easylogging++.cc.o
CMakeFiles/Makefile2:339: recipe for target 'src/gl/CMakeFiles/realsense2-gl.dir/all' failed
make[1]: *** [src/gl/CMakeFiles/realsense2-gl.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

VanKush, 2022-04-07:
[image]

MartyG-RealSense, 2022-04-07:
Does the error still occur if you run the install instruction below and then build?

**sudo apt-get install libglfw3-dev**

VanKush, 2022-04-08:
It always failed at the time of Recompile and install librealsense binaries:

sudo make uninstall && make clean && make && sudo make install make command 
  ^
[ 55%] Building C object src/gl/CMakeFiles/realsense2-gl.dir/__/__/third-party/realsense-file/lz4/lz4.c.o
[ 55%] Building CXX object src/gl/CMakeFiles/realsense2-gl.dir/__/__/third-party/easyloggingpp/src/easylogging++.cc.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libGL.so', needed by 'src/gl/librealsense2-gl.so.2.50.0'.  Stop.
CMakeFiles/Makefile2:339: recipe for target 'src/gl/CMakeFiles/realsense2-gl.dir/all' failed
make[1]: *** [src/gl/CMakeFiles/realsense2-gl.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

VanKush, 2022-04-08:
Finally I have installed it with the instruction given on below link:
 https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md 

Installing the packages:

MartyG-RealSense, 2022-04-08:
I have not seen these errors in a librealsense case before, though the link below that discusses these errors suggests that if a **proxy server** is being used, closing and exiting it may allow installation to proceed normally.

https://www.dev2qa.com/how-to-fix-valueerror-check_hostname-requires-server_hostname/

The link below advises about how to turn off the proxy.

https://superuser.com/questions/196166/linux-command-line-to-turn-off-proxy

MartyG-RealSense, 2022-04-14:
Hi @VanKush  Do you require further assistance with this case, please?  Thanks!

VanKush, 2022-04-15:
No Thank you.

MartyG-RealSense, 2022-04-15:
Thanks very much @VanKush for the update!  As you do not require further assistance, I will close this case. Thanks again!

closed 2022-04-15 by MartyG-RealSense