realsenseai/librealsense #10376: Unable to load JSON parameter file with pyrealsense2

JeffR1992, 2022-04-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.13.00.50) | 
| Operating System & Version |   Windows 10 version 10.0.18363 Build 18363  | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0.3785                          | 
| Language                            |  Python                          | 

### Issue Description
I'm written a simple Python script to record data coming from a D435i into a bag file. The script works fine, however I'm now trying to add additional functionality such as being able to load in a json parameter file before I start the recording. I've looked at the following issue [here](https://github.com/IntelRealSense/librealsense/issues/6075) for some guidance on how to load json files, but unfortunately this did not solve my problem, since I get the following error when running the script:
```

Traceback (most recent call last):
  File "c:\Users\asr\Desktop\depth_camera_recorder.py", line 62, in <module>
    main()
  File "c:\Users\asr\Desktop\depth_camera_recorder.py", line 33, in main
    advnc_mode.load_json(json_string)
RuntimeError: null pointer passed for argument "dev"
```

The code I'm using to try and load the json file and record the data is the following:

```
import pyrealsense2 as rs
import json
import numpy as np
import cv2
import time

def main():
    config_file = json.load(open("C:/Users/asr/Desktop/d435i_test_config.json"))
    json_string = str(config_file).replace("'", '\"')

    h_res = 848
    v_res = 480
    framerate = 30

    pipeline = rs.pipeline()
    config = rs.config()

    # Configure depth and color streams
    config.enable_stream(rs.stream.depth, h_res, v_res, rs.format.z16, framerate)
    config.enable_stream(rs.stream.infrared, h_res, v_res, rs.format.y8, framerate)

    file_path = "C:/Users/asr/Downloads/"
    date_and_time = time.strftime("%Y%m%d-%H%M%S")
    file_name = "depth_camera_output_" + date_and_time + ".bag"
    print(file_path + file_name)
    config.enable_record_to_file(file_path + file_name)

    # pipeline.start(config)
    cfg = pipeline.start(config)
    dev = cfg.get_device()
    advnc_mode = rs.rs400_advanced_mode(dev)
    advnc_mode.load_json(json_string)

    while True:
        frames = pipeline.wait_for_frames()

        infrared_frame = frames.get_infrared_frame()
        infrared_image = np.asanyarray(infrared_frame.get_data())
        infrared_image = np.dstack((infrared_image, infrared_image, infrared_image))

        depth_frame = frames.get_depth_frame()
        depth_image = np.asanyarray(depth_frame.get_data())
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        images = np.hstack((infrared_image, depth_colormap))

        cv2.namedWindow("Video Stream", cv2.WINDOW_AUTOSIZE)
        cv2.imshow("Video Stream", images)

        escape_key = cv2.waitKey(1)
        if escape_key == 27:
            break

if __name__ == "__main__":
    main()
```

MartyG-RealSense, 2022-04-09:
Hi @JeffR1992  To load a json from a string in Python, could you please try using the **serializable.device(dev)** instruction, as described in an example of code at https://github.com/IntelRealSense/librealsense/issues/6902#issuecomment-781364057

JeffR1992, 2022-04-14:
Thanks @MartyG-RealSense, that solved the issue. However it introduced another issue in that I no longer have the following information to work with:
```

"stream-fps": "30",
"stream-height": "480",
"stream-width": "848"
```

so I'm forced to hard code it. This isn't ideal if I want to test a lot of different resolutions and framerate combinations.

Do you know why this error only happens on some computers? Is there a better solution that doesn't remove any of the json information? Thanks.

MartyG-RealSense, 2022-04-14:
Setting the stream resolution and FPS in a json does not usually have a useful effect, as the definition for these settings in the file is overridden when the pipeline is started.  This is why the correct approach is to define the stream settings within the script using **cfg.enable_stream** instructions.

A couple of RealSense users have used a workaround though to load values for the cfg instructions from the json file, as demonstrated in s Python script at https://github.com/IntelRealSense/librealsense/issues/6075

I do not have information about why one PC might accept the json and the other reject it.  Does the problem with loading your json on a particular computer still occur if you edit down the device information to just state the general camera series?

```
{
    "device": {
        "product line": "D400"
    },
```

MartyG-RealSense, 2022-04-19:
Hi @JeffR1992  Do you require further assistance with this case, please?  Thanks!

JeffR1992, 2022-04-21:
Hi @MartyG-RealSense, thanks for your help. You're right, setting the resolution and framerate with `config.enable_stream()` is a better way to go.

I tested the program when loading in a json file that had edited down the device information to just state the general camera series, and I unfortunately I still get the device key error.

MartyG-RealSense, 2022-04-21:
I was able to create a json file without the "device" section that successfully loaded into the Viewer if I made the following edit:

[image]

So that the top of the json file looked like this:

```
{
"parameters": {
```

No changes were made to any other parts of the json such as the bottom section.

Here is my complete edited json that worked.

```
{
    "parameters": {
        "aux-param-autoexposure-setpoint": "400",
        "aux-param-colorcorrection1": "-0.493164",
        "aux-param-colorcorrection10": "-0.272461",
        "aux-param-colorcorrection11": "-0.272461",
        "aux-param-colorcorrection12": "-0.355469",
        "aux-param-colorcorrection2": "0.831055",
        "aux-param-colorcorrection3": "0.831055",
        "aux-param-colorcorrection4": "-0.368164",
        "aux-param-colorcorrection5": "-0.133789",
        "aux-param-colorcorrection6": "-0.323242",
        "aux-param-colorcorrection7": "-0.323242",
        "aux-param-colorcorrection8": "1.19141",
        "aux-param-colorcorrection9": "0.90918",
        "aux-param-depthclampmax": "65536",
        "aux-param-depthclampmin": "0",
        "aux-param-disparityshift": "0",
        "controls-autoexposure-auto": "True",
        "controls-autoexposure-manual": "33000",
        "controls-color-autoexposure-auto": "True",
        "controls-color-autoexposure-manual": "156",
        "controls-color-backlight-compensation": "0",
        "controls-color-brightness": "0",
        "controls-color-contrast": "50",
        "controls-color-gain": "64",
        "controls-color-gamma": "300",
        "controls-color-hue": "0",
        "controls-color-power-line-frequency": "3",
        "controls-color-saturation": "64",
        "controls-color-sharpness": "50",
        "controls-color-white-balance-auto": "True",
        "controls-color-white-balance-manual": "4600",
        "controls-depth-gain": "16",
        "controls-laserpower": "150",
        "controls-laserstate": "on",
        "ignoreSAD": "0",
        "param-amplitude-factor": "0",
        "param-autoexposure-setpoint": "400",
        "param-censusenablereg-udiameter": "9",
        "param-censusenablereg-vdiameter": "9",
        "param-censususize": "9",
        "param-censusvsize": "9",
        "param-depthclampmax": "65536",
        "param-depthclampmin": "0",
        "param-depthunits": "1000",
        "param-disableraucolor": "0",
        "param-disablesadcolor": "0",
        "param-disablesadnormalize": "0",
        "param-disablesloleftcolor": "0",
        "param-disableslorightcolor": "0",
        "param-disparitymode": "0",
        "param-disparityshift": "0",
        "param-lambdaad": "800",
        "param-lambdacensus": "26",
        "param-leftrightthreshold": "24",
        "param-maxscorethreshb": "2047",
        "param-medianthreshold": "500",
        "param-minscorethresha": "1",
        "param-neighborthresh": "7",
        "param-raumine": "1",
        "param-rauminn": "1",
        "param-rauminnssum": "3",
        "param-raumins": "1",
        "param-rauminw": "1",
        "param-rauminwesum": "3",
        "param-regioncolorthresholdb": "0.0499022",
        "param-regioncolorthresholdg": "0.0499022",
        "param-regioncolorthresholdr": "0.0499022",
        "param-regionshrinku": "3",
        "param-regionshrinkv": "1",
        "param-robbinsmonrodecrement": "10",
        "param-robbinsmonroincrement": "10",
        "param-rsmdiffthreshold": "4",
        "param-rsmrauslodiffthreshold": "1",
        "param-rsmremovethreshold": "0.375",
        "param-scanlineedgetaub": "72",
        "param-scanlineedgetaug": "72",
        "param-scanlineedgetaur": "72",
        "param-scanlinep1": "60",
        "param-scanlinep1onediscon": "105",
        "param-scanlinep1twodiscon": "70",
        "param-scanlinep2": "342",
        "param-scanlinep2onediscon": "190",
        "param-scanlinep2twodiscon": "130",
        "param-secondpeakdelta": "325",
        "param-texturecountthresh": "0",
        "param-texturedifferencethresh": "0",
        "param-usersm": "1",
        "param-zunits": "1000"
    },
    "schema version": 1,
    "viewer": {
        "stream-depth-format": "Z16",
        "stream-fps": "30",
        "stream-height": "480",
        "stream-width": "848"
    }
}
```

closed 2022-05-04 by MartyG-RealSense