realsenseai/librealsense #10378: Asic Temperature value is not valid!

Xue21, 2022-04-09:
ERROR [546459083136] (ds5-options.cpp:88) Asic Temperature value is not valid!

MartyG-RealSense, 2022-04-09:
Hi @Xue21  How often does this error occur, and does it stop your program from working when it occurs or does the program continue running normally?

Xue21, 2022-04-10:
Hi @MartyG-RealSense ,It often occurs,but the program continue running normally.
roslaunch realsense2_camera my_vins_d435.launch,then it has WARN as below.

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:35239/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 240
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 424
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 240
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 424
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 240
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 424
 * /camera/realsense2_camera/gyro_fps: 200
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 90 lines cut ...]
[ INFO] [1649555946.291167014]: accel stream is enabled - fps: 100
[ INFO] [1649555946.291233870]: setupPublishers...
[ INFO] [1649555946.302131296]: Expected frequency for depth = 30.00000
[ INFO] [1649555946.439676369]: Expected frequency for infra1 = 30.00000
[ INFO] [1649555946.553715406]: Expected frequency for infra2 = 30.00000
[ INFO] [1649555946.734570853]: Expected frequency for color = 30.00000
[ INFO] [1649555946.859536984]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1649555946.993001239]: Start publisher IMU
[ INFO] [1649555947.006903822]: setupStreams...
[ INFO] [1649555947.065218776]: insert Depth to Stereo Module
[ INFO] [1649555947.065448340]: insert Color to RGB Camera
[ INFO] [1649555947.065564482]: insert Infrared to Stereo Module
[ INFO] [1649555947.065676399]: insert Infrared to Stereo Module
[ INFO] [1649555947.065805087]: insert Gyro to Motion Module
[ INFO] [1649555947.065902378]: insert Accel to Motion Module
 10/04 09:59:07,086 WARNING [546963059072] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 10/04 09:59:07,121 WARNING [546534584704] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/3610000.xhci/usb2/2-3/2-3.1/2-3.1:1.5/0003:8086:0B3A.0004/HID-SENSOR-200073.1.auto/iio:device1/buffer/enable
[ WARN] [1649555947.128033504]: 
[ INFO] [1649555947.232450881]: SELECTED BASE:Depth, 0
[ WARN] [1649555947.285210804]: Hardware Notification:Motion Module failure,1.64956e+12,Error,Hardware Error
[ INFO] [1649555947.381677501]: RealSense Node Is Up!
 10/04 10:00:52,507 ERROR [545922212224] (ds5-options.cpp:88) Asic Temperature value is not valid!
 10/04 10:00:53,509 ERROR [545922212224] (ds5-options.cpp:88) Asic Temperature value is not valid!
 10/04 10:00:54,510 ERROR [545922212224] (ds5-options.cpp:88) Asic Temperature value is not valid!
 10/04 10:02:52,666 ERROR [545922212224] (ds5-options.cpp:88) Asic Temperature value is not valid!

MartyG-RealSense, 2022-04-10:
If your program is running normally then you can likely safely ignore the warning.  The **Asic Temperature value is not valid!** message may occur if the camera is queried for its current temperature in a moment in time when it is not ready to reply.

MartyG-RealSense, 2022-04-22:
Case closed due to no further comments received.

AustinDeric, 2022-11-03:
> If your program is running normally then you can likely safely ignore the warning

@MartyG-RealSense We get this log spam as well which fills up cloud databases and storage space which has costs at scale.  It also makes parsing the logs take longer. 

>  The Asic Temperature value is not valid! message may occur if the camera is queried for its current temperature in a moment in time when it is not ready to reply.

Is this in actual error? If this is an actual error then thats serious and the logic around querying an unready device should be reviewed.  I think its not an error and could be logged at the debug level for devs to figure out why the temperature isn't being published every 10 seconds.

MartyG-RealSense, 2022-11-03:
Hi @AustinDeric  A RealSense team member provides further advice about the **Asic Temperature value is not valid!** error at https://github.com/IntelRealSense/realsense-ros/issues/894#issuecomment-527315705

Although it is labelled by the log as an error, I would categorize it more as a warning that can be ignored if it only occurs occasionally (such as once per minute).  If it occurs frequently then that would increase the potential seriousness of the problem, as the advice in the above link states.

You could potentially deal with log spam by reducing the **logging level**.  Information about setting the logging level in Linux or Windows can be found at the link below.

https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A#q-how-do-i-enable-librealsense-logs

I have re-opened the issue for you.

CXavierh, 2023-07-18:
I also see this error, but i just start the progress, and the camera doesn't feel hot at all. This may be a error of camera.

[ INFO] [1689662607.313302196]: Device with serial number 243122300666 was found.

[ INFO] [1689662607.313344845]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video4 was found.
[ INFO] [1689662607.313362565]: Device with name Intel RealSense D455 was found.
[ INFO] [1689662607.313829379]: Device with port number 2-3 was found.
[ INFO] [1689662607.313852170]: Device USB type: 3.2
[ INFO] [1689662607.316233590]: getParameters...
[ INFO] [1689662607.348177346]: setupDevice...
[ INFO] [1689662607.348197380]: JSON file is not provided
[ INFO] [1689662607.348202023]: ROS Node Namespace: camera
[ INFO] [1689662607.348222563]: Device Name: Intel RealSense D455
[ INFO] [1689662607.348226639]: Device Serial No: 243122300666
[ INFO] [1689662607.348231005]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video4
[ INFO] [1689662607.348238344]: Device FW version: 5.15.0.2
[ INFO] [1689662607.348242325]: Device Product ID: 0x0B5C
[ INFO] [1689662607.348245923]: Enable PointCloud: Off
[ INFO] [1689662607.348249775]: Align Depth: Off
[ INFO] [1689662607.348253410]: Sync Mode: Off
[ INFO] [1689662607.348266393]: Device Sensors: 
[ INFO] [1689662607.351716869]: Stereo Module was found.
[ INFO] [1689662607.356382958]: RGB Camera was found.
[ INFO] [1689662607.356471884]: Motion Module was found.
[ INFO] [1689662607.356486216]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1689662607.356497056]: num_filters: 0
[ INFO] [1689662607.356503644]: Setting Dynamic reconfig parameters.
[ INFO] [1689662607.387435271]: Done Setting Dynamic reconfig parameters.
[ INFO] [1689662607.387782920]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1689662607.388577123]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1689662607.388592808]: gyro stream is enabled - fps: 200
[ INFO] [1689662607.388600251]: accel stream is enabled - fps: 100
[ INFO] [1689662607.388611106]: setupPublishers...
[ INFO] [1689662607.389583878]: Expected frequency for depth = 30.00000
[ INFO] [1689662607.401938555]: Expected frequency for color = 30.00000
[ INFO] [1689662607.407874911]: Start publisher IMU
[ INFO] [1689662607.408144396]: setupStreams...
 18/07 14:43:27,410 WARNING [140137021785856] (ds-calib-parsers.cpp:41) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ INFO] [1689662607.441292821]: SELECTED BASE:Depth, 0
[ INFO] [1689662607.456174207]: RealSense Node Is Up!
[ WARN] [1689662607.522766183]: 
 18/07 14:44:09,450 ERROR [140136771733248] (ds-options.cpp:87) Asic Temperature value is not valid!
 18/07 14:44:09,562 ERROR [140136369080064] (ds-options.cpp:87) Asic Temperature value is not valid!
 18/07 14:44:10,564 ERROR [140136369080064] (ds-options.cpp:87) Asic Temperature value is not valid!
 18/07 14:44:11,450 ERROR [140136771733248] (ds-options.cpp:87) Asic Temperature value is not valid!
 18/07 14:44:11,567 ERROR [140136369080064] (ds-options.cpp:87) Asic Temperature value is not valid!

MartyG-RealSense, 2023-07-18:
Hi @CXavierh  An **Asic Temperature value is not valid!** is not usually a serious problem and is not related to how hot the camera is.  It indicates that when the temperature was requested from the camera hardware, it was not ready to provide the temperature at the moment in time when the temperature value was requested.

Do these errors prevent the launch from running successfully?

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a long time period.

closed 2026-10-23 by MartyG-RealSense