realsenseai/librealsense #10379: Incomplete frame received: Incomplete video frame detected!

Xue21, 2022-04-09:
roslaunch realsense2_camera rs_rgbd.launch
The question is follow.
(backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 508928 out of 1843455 bytes (27%)

MartyG-RealSense, 2022-04-09:
Hi @Xue21  https://github.com/IntelRealSense/realsense-ros/issues/1899#issuecomment-849837579 discusses the **Incomplete frame received: Incomplete video frame detected!** warning in regard to the RealSense ROS wrapper.  

Is any of the advice provided at that link helpful for reducing the occurrence of this warning, please?

Also, are you using an Nvidia Jetson computing board like the RealSense user in that linked-to case?

MartyG-RealSense, 2022-04-22:
Case closed due to no further comments received.

closed 2022-04-22 by MartyG-RealSense

tugbakara, 2022-10-20:
> Also, are you using an Nvidia Jetson computing board like the RealSense user in that linked-to case?

I am getting same warn in ros side is there any issue related to Nvidiav Jetson?

MartyG-RealSense, 2022-10-20:
Hi @tugbakara  The **Incomplete frame received** error is highly rare, and most of the small number of cases that have occured in RealSense ROS are on Jetson boards.

In addition to the resources listed at https://github.com/IntelRealSense/realsense-ros/issues/1899#issuecomment-849837579 further suggestions offered by RealSense ROS users have been:

*  Reduce the FPS to '6' and disable infrared streams if they are enabled.

*  Disable **LPM** in usbcore.

https://www.kernel.org/doc/html/v4.13/driver-api/usb/power-management.html#xhci-hardware-link-pm

tugbakara, 2022-10-20:
Thanks @MartyG-RealSense , I need high fps value and I already disabled infra,so I will try to disable LPM if it is possible. :)