realsenseai/librealsense #10380: Got exception of 'out of range value for argument "x"' after align to Depth

wengjiyao, 2022-04-09:
I have following code in Java for Android to  get the distance at (x, y). If frames are aligned to StreamType.COLOR,  df.getDistance() returns correct distance value. However, If frames are aligned to StreamType.DEPTH, df.getDistance() gets the exception of 'out of range value for argument "x"', even though (x, y) is inside 424x240. Could you please help?

    Align mAlign = new Align(StreamType.DEPTH);
    Runnable mStreaming = new Runnable() {
        @Override
        public void run() {
            try {
                try(FrameSet frames = mPipeline.waitForFrames()) {
                    VideoFrame vf;
                    DepthFrame df;

                    try (FrameSet align = frames.applyFilter(mAlign)) {
                        try (Frame d = align.first(StreamType.DEPTH, StreamFormat.Z16)) {
                            df = d.as(Extension.DEPTH_FRAME);

                            Log.i(TAG, "depth width="+df.getWidth());   // depth width=424
                        }

                        try( Frame c = align.first(StreamType.COLOR, StreamFormat.BGRA8) ){
                            vf = c.as(Extension.VIDEO_FRAME);
                        }

                       // find (x, y) of an object
                       distance = df.getDistance(x, y);
                    }
                }
             }
          }

MartyG-RealSense, 2022-04-10:
Hi @wengjiyao  I have seen this out of range error with an aligned image using get_distance before in a non-Java case (on Python) at https://github.com/IntelRealSense/librealsense/issues/7395 where the error did not occur if alignment was not applied.

Have you defined a specific resolution for the floor stream too?  If not then the program would likely use the default color resolution for the particular RealSense camera model that you are using.  On a 400 Series model such as D415, D435 / D435i or D455, that would be **1280x720**

If you are aligning to StreamType.COLOR then you will be aligning depth to color.  If you use StreamType.DEPTH then you will be aligning color to depth.  Color to depth alignment is far more rarely used than depth to color, and an example application of the technique is the RealSense SDK's **rs-measure** C++ program for measuring the distance between two points on the image.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

The main advantage of color to depth alignment is that if you are using a camera model with different field of view (FOV) sizes for depth and color (D435 / D435i) then the aligned image will be stretched out to full screen size instead of having the edges of the aligned image cut off when performing depth to color alignment.  The edge cut-off happens on D435 / D435i because the color sensor FOV size is smaller than the depth sensor FOV size.

The vast majority of RealSense users use depth to color, and if that provides you with correct results then I would recommend it to you too.

wengjiyao, 2022-04-10:
Hi [MartyG-RealSense](https://github.com/MartyG-RealSense), thank you very much for your reply. If I remove alignment and calculate the coordinate mapping from RGB to Depth myself, seems I get the same error randomly. The resolutions of both RGB and Depth are 424x240. When (x, y) is within the range, I sometimes still get exception of 'out of range value for argument "x"'.

MartyG-RealSense, 2022-04-11:
If the exception occurs randomly instead of all the time when handling mapping yourself, perhaps you could implement en error try-catch to catch the exception and allow the program to continue running instead of exiting.  The RealSense SDK's **java_example** program provides a demonstration of implementing this in Java.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/java_example/app/src/main/java/com/example/realsense_java_example/MainActivity.java#L79-L88

wengjiyao, 2022-04-11:
> If the exception occurs randomly instead of all the time when handling mapping yourself, perhaps you could implement an error **try-catch** to catch the exception and allow the program to continue running instead of exiting. The RealSense SDK's **java_example** program provides a demonstration of implementing this in Java.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/java_example/app/src/main/java/com/example/realsense_java_example/MainActivity.java#L79-L88

That's what I am doing. Do you think if C code will work more stable?

MartyG-RealSense, 2022-04-11:
A different language may not necessarily provide a better outcome.

https://github.com/IntelRealSense/librealsense/issues/6249 is another example of a Java project featuring alignment that had incorrect values from **getDistance(x,y)**.

A RealSense programming expert offers advice in that discussion at https://github.com/IntelRealSense/librealsense/issues/6249#issuecomment-615095311

wengjiyao, 2022-04-11:
Hi [MartyG-RealSense](https://github.com/MartyG-RealSense), thank you very much. It's been very helpful.

closed 2022-04-11 by wengjiyao

MartyG-RealSense, 2022-04-11:
You are very welcome, @wengjiyao  - I'm pleased that I could help  :)