realsenseai/librealsense #10381: u-disparity Map Calculation

hasirk, 2022-04-10:
CameraModel: D435
C++, OpenCV
Ubuntu 20.04

I need to construct the u-disparity map from the disparity values from the stereo image. I get the disparity values as below:

`rs2::depth_frame depth = frames.get_depth_frame();
rs2::disparity_frame disp = depth.apply_filter(depth_to_disparity);

float baseline = disp.get_baseline()/1000.0;
auto width = depth.get_width();
auto height = depth.get_height();

Mat disparityData(Size(width, height), CV_32FC1, (void*)disp.get_data(), Mat::AUTO_STEP);`

When I check the data,disparity values range from 0-2,9XX.XX. However, disparity cannot be more than the width of the image (my current image size is 640x480).

How can I understand these disparity values?

MartyG-RealSense, 2022-04-10:
Hi @hasirk  I researched your question extensively but could not find an existing example of generating a u-disparity map.  In the small number of RealSense disparity script resources that exist, they are typically in Python language instead of C++ and are interested in v-disparity rather than u-disparity.

Your use of **CV_32FC1** catches my attention, as RealSense cameras use **16-bit** values for depth by default.  A RealSense team member suggests code for conversion to CV_32FC1 at https://github.com/IntelRealSense/librealsense/issues/6101#issuecomment-602161932

A C++ script in the link below for generating a raw disparity map with a 400 Series camera may also be of interest.

https://support.intelrealsense.com/hc/en-us/community/posts/360051896473/comments/360014626993

hasirk, 2022-04-10:
Hi @MartyG-RealSense 

Thank you for your reply. I checked the code from the link and it also generate he disparity map and it outputs max value for the disparity is 65535. My code above also generates the disparity image. I think the raw disparity method in the link only get the inverse depth, which is not exactly disparity. I think you should apply the disparity_filter from realsense library to get disparity from depth image.

`rs2::disparity_transform depth_to_disparity(true);`
`rs2::depth_frame depth = frames.get_depth_frame();`
`rs2::disparity_frame disp = depth.apply_filter(depth_to_disparity);`

Actually, the problem I am facing is not generating the disparity image, but how to understand the values.

MartyG-RealSense, 2022-04-11:
The value **65535** corresponds to the maximum raw 16-bit pixel depth value (uint16_t).  The raw depth value has a range of 0-65535 and the real-world distance in meters is obtained by multiplying the pixel depth value by the **depth unit scale** value of the particular RealSense camera model (which is by default 0.001 on most 400 Series models except for the D405).

So 65535 x 0.001 gives a real-world maximum depth distance of 65.535 meters.

When defining **rs2::disparity_transform**, you need to do it as a pair of instructions for both depth_to_disparity and disparity_to_depth, with one true and the other false (they cannot both be true or false).

```
rs2::disparity_transform depth_to_disparity(true);
rs2::disparity_transform disparity_to_depth(false);
```

My understanding is that the disparity value should be 1/32 of the pixel resolution value, as mentioned by RealSense team members in https://github.com/IntelRealSense/librealsense/issues/3344#issuecomment-468171910 and https://github.com/IntelRealSense/librealsense/issues/3039#issuecomment-457773391

hasirk, 2022-04-14:
Thanks again for the reply.

Yes I used the par of instructions and it was my mistake not to mention that.
`rs2::disparity_transform depth_to_disparity(true);`
`rs2::disparity_transform disparity_to_depth(false);`

Thank you for the explanation for depths. So what I assumed is correct that adjusting the depth_table in advance_mode will give the inverse of depth.

As you have mentioned and according to the comments, by dividing `rs2::disparity_frame disp`  I should be able to get the disparity value. I will check that. Thanks a lot.

But do you know the reason why one should divide by '32' ? any particular reason for using a scaled disparity value and why it is 32 ?

MartyG-RealSense, 2022-04-14:
There is not a clear definition available for why 1/32 is used.  My understanding from the limited information available on the subject is that **disparities** is a unit of pixel resolution, and that 1/32 disparities is equal to 1 LSB (Least Significant Bit).  The 
Post-Processing documentation adds that "the RealSense sensors can also output depth directly in disparities which would remove the need for the depth-to-disparity transform".  

The 1/32 disparity value is used with post-processing 'delta' settings with the Temporal and Spatial post-processing filters, with a Delta value of '8' equating to 8/32 **disparities**.

https://dev.intelrealsense.com/docs/depth-post-processing

In regard to the rounding question, I am not certain, though a note in the SDK's code for the Spatial filter that "filtering integer values requires round-up to the nearest discrete value" would seem to indicate that rounding-up is recommended.

https://github.com/IntelRealSense/librealsense/blob/master/src/proc/spatial-filter.h#L66

hasirk, 2022-04-14:
Thank you very much for your reply. 

As per the comment "the RealSense sensors can also output depth directly in disparities which would remove the need for the depth-to-disparity transform", I would be very much interested in this. Do you how I can get this information ? 

The best I could find was to transform depth to disparity to get disparity values. But stereo actually works by first calculating disparity and then calculate depth based on disparity,baseline and focal length. I actually want to use disparity values to generate u-disparity and v-disparity values.

MartyG-RealSense, 2022-04-14:
RealSense team members provide advice about this documentation quote and how to activate this **disparity mode** in C++ at https://github.com/IntelRealSense/librealsense/issues/3344

MartyG-RealSense, 2022-04-20:
Hi @hasirk  Do you require further assistance with this case, please?  Thanks!

hasirk, 2022-04-21:
Hi @MartyG-RealSense 
Thanks for the support. I was already using the method you specified to work on **disparity mode**.
I will close the issue.It solved my problem. 
Thank you.

closed 2022-04-21 by hasirk

MartyG-RealSense, 2022-04-21:
You are very welcome, @hasirk  - thanks for the update!