realsenseai/librealsense #10385: About calibration

HyeYumKim, 2022-04-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello, I'm going to measure the volume of the box with python.
As far as I know, calibration is used a lot to correct distortion. In box_dimensioner_multicam_demo.py provided in python sdk, I would like to know whether calibration is used to correct distortion or to match the actual coordinates. Does the realsense camera not need to correct distortion?
Also, I would like to know if I can get up to 3D points using only stereo principle without depth sensor.
thank you.

MartyG-RealSense, 2022-04-12:
When referring to distortion, if you mean effects that negatively affect the image such as a flat wall being curved inwards or outwards: then yes, performing a calibration can help to correct distortion.  

A calibration is stored inside the camera hardware as a table of intrinsic and extrinsic values.  You do not need to perform calibration when using the camera outside of the RealSense SDK unless there is actually a problem with the image.

Calibrating to a chessboard is used to establish that observed chessboard's pose (position and rotation) in relation to the camera's position and viewpoint.

HyeYumKim, 2022-04-13:
Thank you. Another question, when I measure the volume with box_dimensioner_multicam_demo.py, can you measure the moving object exactly?
If it is harder than static objects, I would like to know if there is a way to improve it.

MartyG-RealSense, 2022-04-13:
box_dimensioner_multicam is not suited to measuring the volume of moving objects.

You could check whether a user-created Python conversion of the SDK program **rs-measure** for measuring between two points instead of between the camera and a point could suit your reauirements.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

It would also likely be helpful to be using a faster FPS rate than normal when checking a moving object.  The RealSense SDK supports 300 FPS with 848x100 resolution on 400 Series camera models with a global shutter on the depth sensor (i.e everything but D415) where the viewpoint is limited to a horizontal area at the center of the screen rather than the whole screen.  If you need a large viewpoint then you can use 848x480 at 90 FPS.

https://dev.intelrealsense.com/docs/high-speed-capture-mode-of-intel-realsense-depth-camera-d435

[image]

closed 2022-04-25 by MartyG-RealSense