realsenseai/librealsense #10387: Communication problem on custom hardware - stuck on wait_for_frames

LadaOndris, 2022-04-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Linux  | 
| Kernel Version (Linux Only)    |  5.4.129-0+git.92246cb6d153                                         | 
| Platform                                 | Custom (Raspberry-like)  |
| SDK Version                            |  2.50.0                           | 
| Language                            |  C++                          | 
| Segment			|  biometrics                          | 

### Issue Description

I am trying to run the D435 camera on a custom Raspberry-like device, but no frame is being received. It is stuck on wait_for_frames.

- I cross-compiled the RealSense library from source, which created **librealsense2.so.2.50.0**
- I cross-compiled the basic example rs-hello-realsense https://dev.intelrealsense.com/docs/rs-hello-realsense
- I copied both the **librealsense2.so.2.50.0** and the compiled basic example onto the target device
- I tried to run the program, but it is stuck on wait_for_frames.

Other notes:
- The camera is being detected (as per dmesg messages).
- I once received a single frame, but only once.
- A powered USB hub did not help in resolving the issue.

I included logs both with a USB hub and without it:

### Logs (without USB hub)

#### dmesg 

```
root@verca:~# ./rs-hello-realsense 
[  134.775091] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  134.782652] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  134.792852] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/32f10108.usb/38200000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1.1/2-1.1:1.0/input/input4
[  134.904579] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  134.912165] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  134.922866] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/32f10108.usb/38200000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1.1/2-1.1:1.0/input/input5
[  134.957424] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[  134.964948] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  137.359256] usb 2-1.1: USB disconnect, device number 3
[  137.584728] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[  137.909503] usb 2-1.1: new SuperSpeed Gen 1 USB device number 4 using xhci-hcd
[  137.953089] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  137.960607] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  137.970273] uvcvideo: Failed to set UVC probe control : -32 (exp. 48).
[  137.977200] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/32f10108.usb/38200000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1.1/2-1.1:1.0/input/input6
[  137.991743] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[  137.999114] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  152.568939] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  152.576756] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  152.587543] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/32f10108.usb/38200000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1.1/2-1.1:1.0/input/input7
[  152.696485] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  152.704387] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  152.716046] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/32f10108.usb/38200000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1.1/2-1.1:1.0/input/input8
[  152.758815] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[  152.766372] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  174.026415] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  174.034076] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  174.043082] uvcvideo: Failed to query (GET_INFO) UVC control 4 on unit 2: -71 (exp. 1).
[  174.051435] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[  174.060050] uvcvideo: Failed to query (129) UVC probe control : -71 (exp. 48).
[  174.067305] uvcvideo: Failed to initialize the device (-5).
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x3f4e5db0):
    Frame didn't arrive within 15000
```

[... 281 lines cut ...]
 12/02 08:03:27,738 INFO [281473109019088] (context.cpp:382) Found 1 RealSense devices (mask 0xfe)
 12/02 08:03:27,738 DEBUG [281473109019088] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-1.1.2-32
 12/02 08:03:27,922 DEBUG [281473109019088] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-1.1.2-32
 12/02 08:03:28,160 DEBUG [281473109019088] (frame-archive.h:202) All frames from stream 0x0x148d75e0 are now released by the user
 12/02 08:03:28,208 INFO [281473109019088] (sensor.cpp:1474) Request: Z16 Depth, 
Resolved to: Z16 Depth, 
 12/02 08:03:28,236 INFO [281473109019088] (uvc-streamer.cpp:28) endpoint 130 read buffer size: 615424
 12/02 08:03:28,243 DEBUG [281473109019088] (global_timestamp_reader.cpp:176) time_diff_keeper::start: _users_count = 1
 12/02 08:03:28,345 WARNING [281472775156112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/02 08:03:28,496 WARNING [281472775156112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/02 08:03:28,648 WARNING [281472775156112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/02 08:03:28,800 WARNING [281472775156112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/02 08:03:28,851 WARNING [281472775156112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/02 08:03:28,903 WARNING [281472775156112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/02 08:03:28,955 WARNING [281472775156112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/02 08:03:29,007 WARNING [281472775156112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/02 08:03:29,059 WARNING [281472775156112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/02 08:03:29,110 WARNING [281472775156112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 12/02 08:03:29,161 WARNING [281472775156112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

I have tried figuring it out by analyzing the source code that produces these errors but without luck.
**What might cause these warnings and errors?**

Feel free to ask me for more information.

MartyG-RealSense, 2022-04-11:
Hi @LadaOndris  I do not think that there is necessarily a problem with your Pi-like hardware or your RealSense setup.  The RealSense SDK example programs such as **rs-hello-realsense** are built in a different way to user-created programs, and I have seen this kind of issue occurring before when RealSense users have tried to build the rs-hello-realsense code with their own method.

One RealSense user who experienced this problem shared a guide that they created for building rs-hello-realsense at https://github.com/IntelRealSense/librealsense/issues/6797

LadaOndris, 2022-04-12:
Thank you, @MartyG-RealSense. I don't think the problem is during compilation. The compilation finishes without errors. For completeness, this is the **CMakeLists.txt** that I use for the rs-hello-realsense example.
```
cmake_minimum_required(VERSION 3.1.0)
project(hello-realsense)

set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED TRUE)

set(realsense2_DIR "/home/lada/downloads/librealsense-master/build")

add_executable(hello-realsense rs-hello-realsense.cpp)
set_property(TARGET hello-realsense PROPERTY CXX_STANDARD 11)

find_package(realsense2 REQUIRED)

target_link_libraries(hello-realsense ${realsense2_LIBRARY})
```

The camera repeatedly reconnects for some unknown reason, and no frame ever comes. 

I failed to find the source of the problem from the log messages.

LadaOndris, 2022-04-12:
I have good news! So I searched for people having problems with camera disconnecting and found out that a **USB cable** might be the source of this issue from the following comment https://github.com/IntelRealSense/librealsense/issues/3333#issuecomment-548914868.  

I **switched** the original RealSense cable **to a cable** from the power adapter of my smartphone, HONOR SuperCharge, and it works!

It would be best if we could determine the exact specifications of the cable that are required for this to work.

MartyG-RealSense, 2022-04-12:
That's excellent news that it's working for you  :)

The cables supplied with RealSense cameras are high quality ones that are validated by Intel for use with RealSense.  Whilst it is highly rare that problems with them are reported, even the best USB cable is not completely immune to problems.

The recommended USB cable specification is listed in the top half of page 118 of the current edition of the data sheet document for the 400 Series cameras.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

USB cables supplied by the company **Newnex** are a suitable replacement for the official cable.

https://www.newnex.com/realsense-3d-camera-connectivity.php

LadaOndris, 2022-04-13:
Thank you, @MartyG-RealSense, for the additional information. Now that the issue is resolved, I think we can close it.

MartyG-RealSense, 2022-04-13:
Okay, thanks very much @LadaOndris - as you suggest, i will close the issue.  Thanks again!

closed 2022-04-13 by MartyG-RealSense