realsenseai/librealsense #10390: realsense world coordinate extract

YongSeongLee25, 2022-04-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (05.13.00.50) | 
| Operating System & Version |   {Ubuntu 18.04}  | 
| Platform                                 | PC |
| Language                            |  {python }                          | 

### Issue Description

https://github.com/gregorsamsa183/RealTime3DPoseTracker-OpenPose/blob/master/createTrainingSet.py

vtx = np.asanyarray(points.get_vertices())

Is that code correct to extract world coordinates?

MartyG-RealSense, 2022-04-12:
Hi @YongSeongLee25   Yes, **points.get_vertices** can be used with a point cloud generated with **pc.calculate** to store 3D xyz world coordinates in a numpy (np) array so that they can then be retrieved from the array.

https://github.com/gregorsamsa183/RealTime3DPoseTracker-OpenPose/blob/master/createTrainingSet.py#L215-L218

YongSeongLee25, 2022-04-12:
> 안녕하세요 @YongSeongLee25 예, **points.get_vertices 는 ****pc.calculate** 로 생성된 포인트 클라우드와 함께 사용하여 3D xyz 세계 좌표를 numpy(np) 배열에 저장하여 배열에서 검색할 수 있습니다.
> 
> https://github.com/gregorsamsa183/RealTime3DPoseTracker-OpenPose/blob/master/createTrainingSet.py#L215-L218

thank!

closed 2022-04-12 by YongSeongLee25