realsenseai/librealsense #10391: The origin point of the point cloud coordinate system

oscarWlde, 2022-04-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (05.13.00.50) | 
| Operating System & Version |   {Win 10 }  | 
| Platform                                 | PC  |
| SDK Version                            | 2                         | 
| Language                            |  python }                          | 
| Segment			|  {location }                          | 

### Issue Description
“Each stream of images provided by this SDK is also associated with a separate 3D coordinate space, specified in meters, with **the coordinate [0,0,0] referring to the center of the physical imager**. Within this space, the positive x-axis points to the right, the positive y-axis points down, and the positive z-axis points forward. Coordinates within this space are referred to as "points", and are used to describe locations within 3D space that might be visible within a particular image.”

I can understand that the point cloud coordinates (0,0,0) correspond to the midpoint of the physical world D455 left imager and right imager, right?

oscarWlde, 2022-04-12:
> Thanks！
I guess you mean left imager ，not  left IR imager.There is just one
IR projector.

MartyG-RealSense, 2022-04-12:
The left and right imagers are both IR sensors.  The projector - also known as the **IR Emitter** - is a separate component from the left and right imagers that is used to project a semi-random dot pattern onto the observed scene in order to help the camera analyze surfaces for depth information.  Disabling the projector therefore does not disable the IR imagers.

The camera hardware generates the depth frame from raw left and right infrared frames.  Because both left and right are used, this is why the camera is called a Stereo depth camera.  These raw frames are not the same as the Infrared and Infrared 2 streams, and so depth is still provided even if Infrared and Infrared 2 are inactive.

closed 2022-04-13 by oscarWlde