realsenseai/librealsense #10392: L515 gives wrong 3D coordinates after running for a while, and the problem will be fixed briefly after plugging again.

weizihao30, 2022-04-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu 18}  | 
| Kernel Version (Linux Only)    |  (5.4.0-107-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.50.0 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I have several L515s in use, and I use the pyrealsense2 library APIs to get frames from L515. The work I'm doing is to align color frames and depth frames, then use the pyrealsense2.rs2_deproject_pixel_to_point() API to calculate the 3D coordinates of a wanted pixel in color frame.
The L515 works good for more than 90% of time, but it goes wrong after using for half a day or longer. Sometimes it appears "frames didn't arrive in 5000" error, sometimes it gives 3D coordinates deviating for more than 70cm. 
Restarting my code doesn't work, while plugging again can fix the problem for a while.

weizihao30, 2022-04-12:
Besides, this problem occurs only on this device, other L515s work well all the time with same python code.

MartyG-RealSense, 2022-04-12:
Hi @weizihao30  When a RealSense camera has been working normally in a long-run session for a prolonged period of time and then suddenly has a problem, it is more likely to be related to an issue other than the camera hardware, such as a glitch in the USB system (the port or the cable), or a problem with the computer hardware or its operating system.  

Are all of the cameras attached to the same computer and active simultaneously?  If they are, what is the hardware specification of the computer, please?  For four simultaneous cameras, a computer with an **Intel i7** processor or equivalent capability would be recommended.  The more cameras that are attached to the same computer, the more of the computer's resources that will be consumed.  In past cases with a 4 camera setup, 3 have sometimes worked fine and the 4th camera has been the one that always experienced a problem.

It would also be worth checking the temperature of the 4th camera at the time that the problem occurs before you unplug it to see whether it is hot.  If it is not possible to use the RealSense Viewer program to check the temperature then touching the casing of that particular camera should enable you to tell whether it is hot to the touch.

MartyG-RealSense, 2022-04-25:
Case closed due to no further comments received.

closed 2022-04-25 by MartyG-RealSense

ADCring, 2022-07-11:
I'm @FeiSong123  co-worker. Yes, the camera is hot, shall I use a fan to help it cool down?

MartyG-RealSense, 2022-07-11:
If the camera casing is hot to the touch after only a few minutes and the air temperature in the location you are using the camera in is relatively cool then it would suggest that the camera is overheating.  This may occur due to an electrical problem with the USB port or the USB cable (for example, due to internal damage within the cable).  Is it possible that the cable could have been damaged it during the robot's fall? 

Directing a fan onto the camera would not be advisable as it may disrupt the air flow around the camera's ventilation grilles and make overheating worse.

MartyG-RealSense, 2022-07-11:
No, wind from the fan would not loosen components inside the camera.  So long as you are not directing the fan's air directly into the ventilation grilles of the camera at close range then it should be fine to have a fan running near the camera.

MartyG-RealSense, 2022-09-05:
The L515 datasheet recommends a maximum casing ('skin') temperature of 50 degrees C, yes.  

Lab tests by Intel on the 400 Series camera models established that there is a relationship between the outer casing temperature and the inner operating temperature.  It was found that if the outer casing temperature did not exceed 44 degrees C then the internal operating temperature should not exceed 60 degrees C.

Whilst a minimum temperature for the casing is not defined in the data sheet, the aim would be to not let its temperature fall below 0 degrees C if possible.

FeiSong123, 2022-09-05:
Does this mean that we should make it work within the temperature range, if not, we should reduce the temperature or heat it to return to the normal range. Otherwise, there will be data error like before?

MartyG-RealSense, 2022-09-05:
There is less information about the effects of very high temperature on the L515 model than there is about the 400 Series models.  However, as a general rule very high temperature in electronics may cause a short-circuit fault, as described at the link below.

https://en.wikipedia.org/wiki/Overheating_(electricity)

If possible, keep the temperatures within the recommended minimum and maximum ranges.