realsenseai/librealsense #10394: Issue on Jetson TX2 NX when activating IMU

bricemiramont, 2022-04-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50| 
| Operating System & Version |   Ubuntu 18.04 L4T  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                      | 
| Platform                                 | Jetson TX2 NX  |
| SDK Version                            |  2.50.0 }                          | 
| Language                            |  python                          | 
| Segment			|  others                         | 

### Issue Description

I have a python script that used to work with only the dpeth and rgb streams enabled.
I need to have the accel stream to process the position of the camera and when I enable it, Frames do not arrive and I have the error below.
Seems like a udev rule not applied correctly.

Jetson has been flashed with Jetpack 4.6 initially. I have applied the L4T kernel patch, and I compiled the realsense lib per this document: https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation

when I run the script, I have this error :

`Traceback (most recent call last):
  File "test.py", line 20, in <module>
    f = p.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000`

syslog has the following errors

`Apr 12 12:07:48 fp-tx2-02 systemd-udevd[9319]: Process '/bin/sh -c ' chmod -R 0777 /sys//devices/3530000.xhci/usb2/2-2/2-2.3/2-2.3:1.5/0003:8086:0B5C.000B/HID-SENSOR-200073.1.auto && chmod 0777 /dev/HID-SENSOR-200073.1.auto'' failed with exit code 1.
Apr 12 12:07:48 fp-tx2-02 systemd-udevd[9379]: Process '/bin/sh -c ' chmod -R 0777 /sys//devices/3530000.xhci/usb2/2-2/2-2.3/2-2.3:1.5/0003:8086:0B5C.000C/HID-SENSOR-200073.1.auto && chmod 0777 /dev/HID-SENSOR-200073.1.auto'' failed with exit code 1.
Apr 12 12:07:48 fp-tx2-02 kernel: [   82.307230] hid-sensor-hub 0003:8086:0B5C.000C: No report with id 0xffffffff found
Apr 12 12:07:48 fp-tx2-02 kernel: [   82.315656] hid-sensor-hub 0003:8086:0B5C.000C: No report with id 0xffffffff found
Apr 12 12:07:48 fp-tx2-02 systemd-udevd[9379]: Process '/bin/sh -c ' chmod -R 0777 /sys//devices/3530000.xhci/usb2/2-2/2-2.3/2-2.3:1.5/0003:8086:0B5C.000C/HID-SENSOR-200073.1.auto && chmod 0777 /dev/HID-SENSOR-200073.1.auto'' failed with exit code 1.
Apr 12 12:07:48 fp-tx2-02 systemd-udevd[9397]: Process '/bin/sh -c ' chmod -R 0777 /sys//devices/3530000.xhci/usb2/2-2/2-2.3/2-2.3:1.5/0003:8086:0B5C.000C/HID-SENSOR-200076.2.auto && chmod 0777 /dev/HID-SENSOR-200076.2.auto'' failed with exit code 1.
Apr 12 12:07:48 fp-tx2-02 kernel: [   82.387245] hid-sensor-hub 0003:8086:0B5C.000C: No report with id 0xffffffff found
Apr 12 12:07:48 fp-tx2-02 kernel: [   82.395635] hid-sensor-hub 0003:8086:0B5C.000C: No report with id 0xffffffff found
Apr 12 12:07:48 fp-tx2-02 kernel: [   82.403448] hid-sensor-hub 0003:8086:0B5C.000C: No report with id 0xffffffff found
Apr 12 12:07:48 fp-tx2-02 kernel: [   82.411098] hid-sensor-hub 0003:8086:0B5C.000C: No report with id 0xffffffff found
Apr 12 12:07:48 fp-tx2-02 kernel: [   82.418963] hid-sensor-hub 0003:8086:0B5C.000C: No report with id 0xffffffff found
Apr 12 12:07:48 fp-tx2-02 kernel: [   82.426898] hid-sensor-hub 0003:8086:0B5C.000C: No report with id 0xffffffff found`

The same python script works in a Ubuntu 18.04 VM running in Vmware Fusion on a Mac (x86)

thank you

MartyG-RealSense, 2022-04-12:
Hi @bricemiramont  It is a known issue that having three stream types (depth, RGB, IMU) enabled at the same time can cause **No Frames Received** problems in the RealSense SDK.  It does not matter which order the streams are enabled in, as enabling the IMU streams is always the trigger.

It does not occur for every RealSense user, which may account for why you do not experience the problem on your Mac VM setup but do on Nvidia Jetson (i.e it is a computer-related issue rather than a camera hardware issue).  

A workaround in Python is to set up two separate pipelines, with IMU on one pipeline and depth & color on the other pipeline. 
 The best example of a Python script for implementing this is at https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238

bricemiramont, 2022-04-12:
Hi Marty, 
ok thank you for the info, I'll try that.

thank you,

Brice

bricemiramont, 2022-04-12:
Hello, 

sorry about that but I have the same issue when only the accel stream is enabled.

bricemiramont, 2022-04-12:
I built the lib from the source code.

I am building it with RSUSB backend with the libuvc_installation script.
I added python bindings and cuda boolean in the sh script and it is compiling at the moment.
I will try the lib after that, with two pipelines (one for imu and one for depth/rgb). I'll keep you informed

MartyG-RealSense, 2022-04-12:
Thanks very much for the update - I look forward to your next report.  Good luck!

bricemiramont, 2022-04-12:
Hello Marty,

It does work with the RSUSB compilation process. I reverted back to one pipeline for the three streams and it does work.

thank you,

Brice

closed 2022-04-12 by bricemiramont

MartyG-RealSense, 2022-04-12:
You are very welcome.  It's great to hear that RSUSB worked for you  :)