realsenseai/librealsense #10395: raspberry pi 4B can not get correct imu data

CsHanWeiQi, 2022-04-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   ubuntu-mate 20.04      | 
| Kernel Version (Linux Only)    |         5.4.0-1058-raspi                    | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  { legacy / 2.5.0 }                          | 
| Language                            |  C++                         | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
When running MSCKF code, IMU data cannot be obtained correctly via rs_camera.launch . When not running the program, the data is OK.
[image]
The figure shows the data obtained when the program is running and the IMU is still

MartyG-RealSense, 2022-04-13:
Hi @CsHanWeiQi  In https://github.com/IntelRealSense/realsense-ros/issues/898#issuecomment-525768272 a RealSense ROS user suggests that MSCKF is a type of algorithm that expects the IMU to be operating at a fixed rate.  

In the ROS wrapper, a method of achieving a steady IMU rate instead of a varying one is to combine the separate gyro and accel topics into a single 'imu' topic with the **unite_imu_method** parameter and configure its mode to **copy** instead of the more common 'linear_interpolation'.

The mode of unite_imu_method can either be defined in the launch file or by including it in the launch instruction.  For example for the ROS1 rs_camera.launch instruction:

**roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=copy**

MartyG-RealSense, 2022-04-22:
Hi @CsHanWeiQi  Do you require further assistance with this case, please?  Thanks!

CsHanWeiQi, 2022-04-22:
Thanks for your reply, but after I make these changes, the IMU on the Raspberry Pi still cannot read the data correctly. Is there a problem with the interface support for ros on the Raspberry Pi?

CsHanWeiQi, 2022-04-22:
Hi @MartyG-RealSense . Thanks for your reply, but after I make these changes, the IMU on the Raspberry Pi still cannot read the data correctly. Is there a problem with the interface support for ros on the Raspberry Pi?

MartyG-RealSense, 2022-04-22:
There have not been many reported problems with IMU topics on RealSense ROS installations on Raspberry Pi, though that may be because it is more common for Pi users to use depth and RGB topics rather than IMU.  

You mentioned at the start of this case that the IMU problems occur when using MSCKF code and the IMU is okay when not using MSCKF.  As far as my research of this case can tell, there are very few RealSense ROS users who use MSCKF and so there are virtually no RealSense ROS related reference materials on the subject.

 I wonder whether the wrapper launch file **opensource_tracking.launch** would be a better fit for an IMU related project than **rs_camera.launch**

**roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true unite_imu_method:=copy**

closed 2022-05-04 by MartyG-RealSense