realsenseai/librealsense #10397: How to use rs.option.color_scheme to set withe to black mode

JC-FOSU, 2022-04-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.0 | 
| Language                            |  python  | 
| Segment			|  {others }                          | 

### Issue Description
WINDOWS10

MartyG-RealSense, 2022-04-14:
Hi @JC-FOSU  Resources for Python color scheme and color preset programming can be found at https://github.com/IntelRealSense/librealsense/issues/7767#issuecomment-726704671

The color scheme number for White to Black is '2'.

JC-FOSU, 2022-04-14:
I tried using python, opencv and pyrealsense to generate a white to black depth map, but I only found rs.option.color_scheme and I am confused about it.

> '2'.

Thank you, I also want to ask how to use rs.l500_visual_preset, or if there is any reference to use it.

MartyG-RealSense, 2022-04-14:
The RealSense SDK uses a function called a **colorizer** to color-shade depth coordinates according to their values.  For example, the default colorizer setting is to shade from blue to red, with the color coordinates nearest to the camera shaded blue and progressively shifting towards red as distance from the camera increases.

Does the script at https://github.com/IntelRealSense/librealsense/issues/8161#issuecomment-761908864 for setting an L515 visual preset work for you, please?

JC-FOSU, 2022-04-14:
Thank you, my problem has been solved.

MartyG-RealSense, 2022-04-14:
Great to hear, @JC-FOSU - thanks very much for the update!

closed 2022-04-22 by MartyG-RealSense