realsenseai/librealsense #10399: question

yoram7226, 2022-04-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.0  }                          | 
| Language                            |  {C }                          | 
| Segment			|  Robot }                          | 

### Issue Description
I project 3d vertex using  rs2_project_point_to_pixel   and draw the 2d projection point on the image.   The 2d point are shifted by 20 pixels up. Can you tell me what I am doing wrong?  here is my code and the image with the projection 
void
test(rs2::frameset frames, rs2::pipeline pipe )
{
    rs2::pipeline_profile selection = pipe.get_active_profile();
       auto color_stream = selection.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
       rs2_intrinsics e = color_stream.get_intrinsics();

    rs2::video_frame vf = frames.get_depth_frame();

       rs2::pointcloud pc;
       rs2::points points;
       points = pc.calculate(vf);

       //auto Texture_Coord = points.get_texture_coordinates();
       auto Vertex = points.get_vertices();

       for (int i = 0; i < points.size(); i++)
       {
             float other_pixel[3], p[3];
             p[0] = Vertex[i].x;
             p[1] = Vertex[i].y;
             p[2] = Vertex[i].z;

        if( p[0] == 0 || p[1] == 0 || p[2] == 0 )
            continue;

             rs2_project_point_to_pixel(other_pixel, &e, p);

      
        fprintf( stdout, "%f  %f\n", other_pixel[0], other_pixel[1] )

       }
}

[image]

MartyG-RealSense, 2022-04-14:
Hi @yoram7226  These two methods (rs2_deproject_pixel to point / rs2_project_point_to_pixel and points.get_vertices) are not typically mixed together in the same script.  Instead, one method or the other (but not both methods) is used when generating a pointcloud and retrieving the coordinates.

Having said that, there have been rare occasions when vertices have been converted with rs2_project_point_to_pixel.  At https://github.com/IntelRealSense/librealsense/issues/2204#issuecomment-411066051 a RealSense team member suggests code for doing so.

```
float point[2];
float vertex[3] { x, y, z };
rs2_project_point_to_pixel(point, &intrinsics, vertex);
// read output pixel coordinates from point[0] and point[1]
```

yoram7226, 2022-04-14:
Thanks for your fast reply. my target it to find the position of each lidar point on the image. for further investigation.  what is the simple way to do this? to get the exact lidar point  on the image?

MartyG-RealSense, 2022-04-14:
If you only needed a specific coordinate rather than the coordinates of the entire image then you could use the instruction **rs2_project_color_pixel_to_depth_pixel** to convert a single color pixel to a depth pixel.  This instruction should process faster as it does not require aligning the entire image.  

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/2982 https://github.com/IntelRealSense/librealsense/issues/2982 provided a C++ script that used rs2_project_color_pixel_to_depth_pixel to project the color pixel to the depth pixel and then found the value of the depth pixel after that and finally deprojected the depth pixel to a 3D point.

yoram7226, 2022-04-15:
Thanks for your fast reply, I look at #2982 and it is more clear now. two  questions.  
 1. How can I get the confidence of each depth data?  
2.   looking on the following code:
     rs2::pointcloud pc;
    pc.map_to(color_frame);
    rs2::points points;
    points = pc.calculate(depth_frame);
   auto Texture_Coord = points.get_texture_coordinates();
    auto Vertex = points.get_vertices();

  pc.calculate   calculate   3d vertex  and  2d texture-coord.    The texture-cords are irregular grid and they are located well on the color image.   My question is how does the      Texture_Coord  calculate? What is the connection between the vertex and the texture-coord?  Is there document describing this calculation?

MartyG-RealSense, 2022-04-15:
1.  Are you asking how to enable the depth sensor **confidence stream** on L515, please?  If so then you can enable this stream with a cfg instruction.

**cfg.enable_stream(rs2_stream::RS2_STREAM_CONFIDENCE, 640, 480, RS2_FORMAT_RAW8, 30);**

The image below shows the confidence stream enabled in the RealSense Viewer.

[image]

If you want to **read** the confidence values then https://github.com/IntelRealSense/librealsense/issues/7029#issuecomment-734031379 provides C++ code.

2.  I am not personally familiar with the workings of texture coordinates.  So I will instead refer to an explanation provided by a programming expert in the RealSense community, quoted below.

*********

The texture coordinates return uv coordinates mapping to the colour image with the coordinates normalised to 0-1.  So to get from uv to xy coordinates (ie, pixel coordinates in the colour image) you have to multiply the u and v values by the colour image width and height (though that code is slightly more complex to account for the extreme edges of the image).

Alternatively, I believe you can first align the depth image to colour using Align, then pass that aligned depth image to the point cloud to get your point cloud aligned to the colour image if you'd prefer.  Then getting the colour of a vertex or the vertex of a colour pixel is simple as they will have the same indices of their respective arrays.  So (x,y) in colour image has corresponding vertex points[x + y * colourWidth] and vice-versa)

MartyG-RealSense, 2022-04-22:
Hi @yoram7226  Do you require further assistance with this case, please?  Thanks!

yoram7226, 2022-04-22:
Hi, 

Thanks for your help, for now I don’t have problems.  

Thabks’

Yoram

closed 2022-04-22 by MartyG-RealSense