realsenseai/librealsense #10403: 您好，请问PLY文件可以传输到数据库吗

woshicaobin, 2022-04-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2022-04-15:
Translation: Hello, can the PLY file be transferred to the database?

Hello @woshicaobin  Could you provide more information please about what you are attempting to do?  Thanks!

你好@woshicaobin 你能提供更多關於你試圖做什麼的信息嗎？ 謝謝！

woshicaobin, 2022-04-15:
我想要做一个系统，首先得将Realsence远程传输至我的算法中，所以得先传入数据库中，您看能行吗

woshicaobin, 2022-04-15:
我想要做一个系统，首先得将Realsence采集的PLY点云远程传输至我的算法中，所以得先传入数据库中，您看能行吗

MartyG-RealSense, 2022-04-15:
Can you tell me more about your database please?   Thanks very much for your patience!

你能告訴我更多關於你的數據庫嗎？ 非常感謝您的耐心等待！

MartyG-RealSense, 2022-04-22:
Hello @woshicaobin  Do you require further assistance with this case, please?  Thanks!

您好@woshicaobin 請問您在這個案子上需要進一步的幫助嗎？ 謝謝！

MartyG-RealSense, 2022-04-28:
Case closed due to no further comments received.

由於沒有收到進一步的評論，案件已結案。

closed 2022-04-28 by MartyG-RealSense