realsenseai/librealsense #10408: how to get the pose_frame in python

TitansWhale, 2022-04-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L500 | 
| Firmware Version                   | 01.05.08.01| 
| Operating System & Version |   ubuntu 20  | 
| Kernel Version (Linux Only)    |                                    | 
| Platform                                 | PC |
| SDK Version                            |                         | 
| Language                            |  python/unity }                         | 
| Segment			|  other                          | 

### Issue Description
I want to Align the z coordinate with the real horizontal.so I need the imu data to tansform my pointcloud data.I referenced rs-pose.cpp.

my code 
```
import pyrealsense2 as rs

p = rs.pipeline()
cfg = rs.config()

cfg.enable_stream(rs.stream.pose, rs.format.six_dof)
p.start(cfg)

while True:
    f = p.wait_for_frames()
    ff = f.first_or_default(rs.stream.pose)
    print(ff)
```
out put:
```
Traceback (most recent call last):
  File "/home/liux/Desktop/jumpwork/temp.py", line 7, in <module>
    p.start(cfg)
RuntimeError: Couldn't resolve requests

Process finished with exit code 1
```

TitansWhale, 2022-04-16:
rs-pose.cpp
```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2019 Intel Corporation. All Rights Reserved.
#include <librealsense2/rs.hpp>
#include <iostream>
#include <iomanip>
#include "example-utils.hpp"

int main(int argc, char * argv[]) try
{
    std::string serial;
    if (!device_with_streams({ RS2_STREAM_POSE}, serial))
        return EXIT_SUCCESS;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    if (!serial.empty())
        cfg.enable_device(serial);
    // Add pose stream
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
    // Start pipeline with chosen configuration
    pipe.start(cfg);

    // Main loop
    while (true)
    {
        // Wait for the next set of frames from the camera
        auto frames = pipe.wait_for_frames();
        // Get a frame from the pose stream
        auto f = frames.first_or_default(RS2_STREAM_POSE);
        // Cast the frame to pose_frame and get its data
        auto pose_data = f.as<rs2::pose_frame>().get_pose_data();

        // Print the x, y, z values of the translation, relative to initial position
        std::cout << "\r" << "Device Position: " << std::setprecision(3) << std::fixed << pose_data.translation.x << " " <<
            pose_data.translation.y << " " << pose_data.translation.z << " (meters)";
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

MartyG-RealSense, 2022-04-16:
Hi @TitansWhale  The pose stream is only accessible on the **RealSense T265 Tracking Camera** model.  This is reflected in the **RunTimeError: Couldn't resolve requests** error message that you experienced, because pose is not a supported mode on L515 and 400 Series cameras.

As an alternative for establishing a relationship between the camera and the ground, a **plane-fit** algorithm may meet your needs, as described by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/6338#issuecomment-623350905

TitansWhale, 2022-04-16:
> Hi @TitansWhale The pose stream is only accessible on the **RealSense T265 Tracking Camera** model. This is reflected in the **RunTimeError: Couldn't resolve requests** error message that you experienced, because pose is not a supported mode on L515 and 400 Series cameras.
> 
> As an alternative for establishing a relationship between the camera and the ground, a **plane-fit** algorithm may meet your needs, as described by a RealSense team member at [#6338 (comment)](https://github.com/IntelRealSense/librealsense/issues/6338#issuecomment-623350905)

thanks

MartyG-RealSense, 2026-10-23:
This open GitHub issue relating to the RealSense L515 camera is being closed because the L515 model is End of Life (EOL) and no longer supported.  We welcome the creation of new issues relating to supported cameras.

closed 2026-10-23 by MartyG-RealSense