realsenseai/librealsense #10410: Get Disparsity with rs2::software_device

ilyak93, 2022-04-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D465                                                       | 
| Operating System & Version | Win 10                                                    | 
| Platform                                 | PC                                                           |
| SDK Version                           | 2.34.0-0~realsense0.2250                       | 
| Language                               |  C++                                                       | 

Hi, can I somehow retrieve the disparity maps from the depth frames I get from the loaded frames into synthetic stream i.e rs2::software_device ?

The following code doesn't seem to work, although the frames are correct in the software sensors:

```
fs = sync.wait_for_frames();
        if (fs.size() == 2) {

            rs2::frame e_frame = fs.get_depth_frame();

            rs2::disparity_transform disparity_transform(true);
            rs2::disparity_frame disparity_frame = disparity_transform.process(e_frame);
            cv::Mat mDisparity = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)disparity_frame.get_data());
            cv::Mat e_r = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)e_frame.get_data());
            cv::imshow("ddd",mDisparity);
            cv::waitKey(0);
            cv::destroyAllWindows();

            rs2::disparity_transform disparity_transform2(false);
            rs2::disparity_frame disparity_frame2 = disparity_transform.process(disparity_frame);
            cv::Mat rec = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)disparity_frame2.get_data());

            cv::imshow("rec",rec);
            cv::waitKey(0);
            cv::destroyAllWindows()
``` 

The reconstructed depth seems to be similar if no the same as the disparity.

depth:
[image]

disparity:
[image]

reconstructed depth from disparity:
[image]

MartyG-RealSense, 2022-04-17:
Hi @ilyak93  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/5385 attempts to use disparity transform in a software_device script and experiences problems.

A RealSense team member provides programming advice in the case, including advice about using software_device with disparity at https://github.com/IntelRealSense/librealsense/issues/5385#issuecomment-563461073

ilyak93, 2022-04-17:
@MartyG-RealSense 

> Just a wild guest - depth2disparity is applicable in case of stereo-based devices only. And in case of software device the mockup sensor should be properly configured to mimic all the attributes of stereo depth camera : intrinsic/extrinsic/base line/etc. It is possible that one of the attributes is not populated properly. Can you post the device/sensors initialization and configuration segment?
What was the type of camera that you used to generated the input PNG files ?
Also - did you try to apply the filters before alignment, and of so - what were the results?

I wonder what else parameter could be in:

>  mimic all the attributes of stereo depth camera : intrinsic/extrinsic/base line/etc

@ev-mp, maybe you can detail it a bit or look at the issue ?
I get the frames, but the transformation doesn't give the correct disparity.

MartyG-RealSense, 2022-04-17:
@ilyak93  I will research this subject tomorrow.  Thanks very much for your patience in the meantime, and please do update here with any findings that you make in the meantime.  Good luck!

ilyak93, 2022-04-17:
@MartyG-RealSense, thanks a lot. I have found a little what seems to be a bug, but that didn't fix the issue. 

In the posted peace of code:
```
fs = sync.wait_for_frames();
        if (fs.size() == 2) {

            rs2::frame e_frame = fs.get_depth_frame();

            rs2::disparity_transform disparity_transform(true);
            rs2::disparity_frame disparity_frame = disparity_transform.process(e_frame);
            cv::Mat mDisparity = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)disparity_frame.get_data());
            cv::Mat e_r = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)e_frame.get_data());
            cv::imshow("ddd",mDisparity);
            cv::waitKey(0);
            cv::destroyAllWindows();

            rs2::disparity_transform disparity_transform2(false);
            rs2::disparity_frame disparity_frame2 = disparity_transform.process(disparity_frame);
            cv::Mat rec = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)disparity_frame2.get_data());

            cv::imshow("rec",rec);
            cv::waitKey(0);
            cv::destroyAllWindows()
```

When I try to apply the opposite transform to transform the disparity I got (maybe it is correct, who knows) to back again to depth I define it as ` rs2::disparity_frame` which isn't correct, if I need to get depth_frame so I changed it to ` rs2::depth_frame`:
```

fs = sync.wait_for_frames();
        if (fs.size() == 2) {

            rs2::frame e_frame = fs.get_depth_frame();

            rs2::disparity_transform disparity_transform(true);
            rs2::disparity_frame disparity_frame = disparity_transform.process(e_frame);
            cv::Mat mDisparity = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)disparity_frame.get_data());
            cv::Mat e_r = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)e_frame.get_data());
            cv::imshow("ddd",mDisparity);
            cv::waitKey(0);
            cv::destroyAllWindows();

           rs2::disparity_transform disparity_transform2(false);
            rs2::depth_frame depth_frame2 = disparity_transform.process(disparity_frame);
            cv::Mat rec = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)depth_frame2.get_data());

            cv::imshow("rec",rec);
            cv::waitKey(0);
            cv::destroyAllWindows();

            cv::imshow("rec",rec);
            cv::waitKey(0);
            cv::destroyAllWindows()
```

but as I noted. it didn't fix it, the transform wasn't successful.

I'll wait for any additional observation, thank you in advance.

MartyG-RealSense, 2022-04-18:
A RealSense user of C++ at the link below tried a similar approach to yours to get the disparity map and it came out incorrect.

https://support.intelrealsense.com/hc/en-us/community/posts/360051896473-Get-Disparity-Map-of-D455-by-RealSense-SDK-2-0

[image]

They did eventually succeed with a different approach at the end of the case.

https://support.intelrealsense.com/hc/en-us/community/posts/360051896473/comments/360014626993

ilyak93, 2022-04-18:
@MartyG-RealSense, oh my bad, the line after that I used the old one without the '2' that you noticed about, now fixed it and the reconstructed depth is correct, so I guess the disparity is too. 

I actually need to check it, saving live during the recording the disparities in 'Advanced Mode' as in the issue you mentioned and see if later using the transform I get depth which makes sense.

I'll check it a bit later. For now the transformation works one direction and the opposite, which may be an indicator of correct disparity (but for the sake of the case, probably it's better to check it in the way I said).

MartyG-RealSense, 2022-04-18:
Yes, please do feel free to run tests to confirm that the code is working to your satisfaction and let me know the results.  Thanks!

I would think that the results could be judged to be correct if you can toggle repeatedly between depth and disparity and have repeatable results in each mode if the camera is not being moved.

closed 2022-05-04 by MartyG-RealSense