realsenseai/librealsense #10411: How to get camera internal and external parameters with python

YongSeongLee25, 2022-04-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (05.13.00.50) | 
| Operating System & Version |   {Ubuntu 18.04}  | 
| Platform                                 | PC |
| Language                            |  {python }                          | 

[### Issue Description
<Describe your issue / question / feature request / etc..>](https://github.com/IntelRealSense/librealsense/issues/10012#issue-1066978384

I found a way to get camera internal and external parameters in that issue.
Do you know how to import this from python pyrealsense2?

MartyG-RealSense, 2022-04-18:
Hi @YongSeongLee25   https://github.com/IntelRealSense/librealsense/issues/1231#issuecomment-368421888 has an example of Python code shared by a RealSense user for retrieving intrinsics and extrinsics.

YongSeongLee25, 2022-04-18:
> 안녕@YongSeongLee25 [#1231(주석)](https://github.com/IntelRealSense/librealsense/issues/1231#issuecomment-368421888) 에는 내장 및 외장을 검색하기 위해 RealSense 사용자가 공유하는 Python 코드의 예가 있습니다.

thank you!

closed 2022-04-18 by YongSeongLee25