realsenseai/librealsense #10412: Saving a colorful pointcloud using sofware-device

ilyak93, 2022-04-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D465                                                       | 
| Operating System & Version | Win 10                                                    | 
| Platform                                 | PC                                                           |
| SDK Version                           | 2.48.0.3381                                             | 
| Language                               |  C++                                                       | 

I'm creating a software-device in the following way:
```
    rs2::software_device dev;

    auto depth_sensor = dev.add_sensor("Depth");
    auto color_sensor = dev.add_sensor("Color");

    rs2_intrinsics depth_intrinsics{W, H, stof(depth[3]), stof(depth[4]),
                                    stof(depth[5]), stof(depth[6]),
                                    RS2_DISTORTION_BROWN_CONRADY,
                                    {stof(depth[7]), stof(depth[8]),
                                     stof(depth[9]), stof(depth[10]),
                                     stof(depth[11])}};
    rs2_intrinsics color_intrinsics{W, H, stof(color[3]), stof(color[4]),
                                    stof(color[5]), stof(color[6]),
                                    RS2_DISTORTION_INVERSE_BROWN_CONRADY,
                                    {stof(color[7]), stof(color[8]),
                                     stof(color[9]), stof(color[10]),
                                     stof(color[11])}};

    auto depth_stream = depth_sensor.add_video_stream(
            {RS2_STREAM_DEPTH, 0, 0, W, H, 30, 2, RS2_FORMAT_Z16,
             depth_intrinsics});
    auto color_stream = color_sensor.add_video_stream(
            {RS2_STREAM_COLOR, 0, 1, W, H, 30, 3, RS2_FORMAT_RGB8,
             color_intrinsics});

    depth_sensor.add_read_only_option(RS2_OPTION_DEPTH_UNITS, 0.001f);
    depth_sensor.add_read_only_option(RS2_OPTION_STEREO_BASELINE, 0.001f);

    depth_stream.register_extrinsics_to(color_stream,
                                        {
        {
            stof(depth_to_color[1]),stof(depth_to_color[2]),
            stof(depth_to_color[3]),stof(depth_to_color[4]),
            stof(depth_to_color[5]),stof(depth_to_color[6]),
            stof(depth_to_color[7]),stof(depth_to_color[8]),
            stof(depth_to_color[9])
            },
            {
            stof(depth_to_color[11]), stof(depth_to_color[12]),
            stof(depth_to_color[12])
            }
                                        });

    dev.create_matcher(RS2_MATCHER_DEFAULT);
    rs2::syncer sync;

    depth_sensor.open(depth_stream);
    color_sensor.open(color_stream);

    depth_sensor.start(sync);
    color_sensor.start(sync);

    rs2::align align(RS2_STREAM_COLOR);

    rs2::frameset fs;
    rs2::frame depth_frame;
    rs2::frame color_frame;

    cv::Mat color_image;
    cv::Mat depth_image;
```

and while looping over the frames:
```
        string color_image_path = string("G:/Vista_project/finish1/rs/") + string(ent_color.d_name);
        string depth_image_path = string("G:/Vista_project/finish1/rs/") + string(ent_depth.d_name);
        color_image = cv::imread(color_image_path.c_str(), cv::IMREAD_COLOR);
        depth_image = cv::imread(depth_image_path.c_str(),cv::IMREAD_UNCHANGED);

        color_sensor.on_video_frame({(void*) color_image.data, // Frame pixels from capture API
                                     [](void*) {}, // Custom deleter (if required)
                                     3 * 1280, 3, // Stride and Bytes-per-pixel
                                     double(idx * 30), RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME,
                                     idx, // Timestamp, Frame# for potential sync services
                                     color_stream});
        depth_sensor.on_video_frame({(void*) depth_image.data, // Frame pixels from capture API
                                     [](void*) {}, // Custom deleter (if required)
                                     2 * 1280, 2, // Stride and Bytes-per-pixel
                                     double(idx * 30), RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME,
                                     idx, // Timestamp, Frame# for potential sync services
                                     depth_stream});

        fs = sync.wait_for_frames();
        pc_idx++;
        if (fs.size() == 2) {

            rs2::frame depth_frame2 = fs.get_depth_frame();
            auto color_frame2 = fs.get_color_frame();

            rs2::pointcloud pc;
            rs2::points points;
            points = pc.calculate(depth_frame2);
            pc.map_to(color_frame2);
            //const rs2::texture_coordinate* coordinates =  points.get_texture_coordinates();
            points.export_to_ply("G:/Vista_project/finish1/rs-ptc/"
                                 + to_string(pc_idx) + "1.ply", color_frame2);
            rs2::save_to_ply exporter("G:/Vista_project/finish1/rs-ptc/"
                                      + to_string(pc_idx) + ".ply", pc);
            exporter.set_option(rs2::save_to_ply::OPTION_PLY_BINARY, 0.f);
            exporter.set_option(rs2::save_to_ply::OPTION_IGNORE_COLOR, 0.f);
            exporter.export_to_ply(points, color_frame2);

            /*
            rs2::frame e_frame = fs.get_depth_frame();

            rs2::disparity_transform disparity_transform(true);
            rs2::frame disparity_frame = disparity_transform.process(e_frame);
            cv::Mat mDisparity = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)disparity_frame.get_data());
            //cv::Mat e_r = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)e_frame.get_data());
            cv::imshow("ddd",mDisparity);
            cv::waitKey(0);
            cv::destroyAllWindows();

            rs2::disparity_transform disparity_transform2(false);
            rs2::frame depth_frame2 = disparity_transform2.process(disparity_frame);
            cv::Mat rec = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)depth_frame2.get_data());
            //cv::Mat e_r2 = cv::Mat(cv::Size(1280, 720), CV_16UC1, (void*)e_frame.get_data());

            cv::imshow("rec",rec);
            cv::waitKey(0);
            cv::destroyAllWindows();
            */

            fs = align.process(fs);
            rs2::frame depth_frame = fs.get_depth_frame();
            rs2::frame color_frame = fs.get_color_frame();

            cv::Mat aligned_image(720, 1280, CV_16UC1, (void*) (depth_frame.get_data()), 2 * 1280);
            cv::imwrite("G:/Vista_project/finish1//aligned_rs/" + string(ent_color.d_name), aligned_image);

        }
        idx++;
```

I'm trying to save the pointcloud in a ply format.

I try both methods: using `points.export_to_ply` (many people wrote here in issues that it worked for them (probably on live streaming and not software-device)) and the method suggested on issue #6677.

I debugged the function `export_to_ply` of the object `rs2::save_to_ply` and found that `texcoords[i]` for each `i` (all the points) are all set to 0 (the `cout << "here"` loop is my code injected into real sense `rs_export.hpp`, which why probably I get only one color (the color of `(u=0,v=0)`), which means my pointcloud somehow doesn't know what is his texture coordinates. Maybe it is somehow connected to the fact it is software-device, some data maybe lost, I think the implementation of this function `rs2::points points = pc.calculate(depth_frame2);` (calculate) should give the answers if you can look around it.

[image]

Thank you in advance.

@MartyG-RealSense, that's the new issue from the #6677 discussion.

MartyG-RealSense, 2022-04-19:
hI @ilyak93  When **OPTION_PLY_BINARY** is set to 0 / false, it exports the ply as **ASCII** text characters, as described in the save_to_ply documentation at https://github.com/IntelRealSense/librealsense/pull/4906

So can you try using false / true instead of 0 / 1 and setting OPTION_PLY_BINARY to true to see whether it results in the export of a colored cloud, please?

```
exporter.set_option(rs2::save_to_ply::OPTION_PLY_BINARY, true);
exporter.set_option(rs2::save_to_ply::OPTION_IGNORE_COLOR, false);
```

ilyak93, 2022-04-19:
@MartyG-RealSense,
I have found a solution in issue #900 , this have worked for me:

> 

> Hi @tchapi
> 
> Can you try:
> 
> ```c++
> auto color = frames.get_color_frame();
> pc.map_to(color);
> points = pc.calculate(depth);
> ```
> 
> Is the `rs-pointcloud` demo working properly on your system?

closed 2022-04-19 by ilyak93

MartyG-RealSense, 2022-04-19:
That's great to hear!  I was just going to show you an old case at https://github.com/IntelRealSense/librealsense/issues/900 that was experiencing (0,0) texcoords.  A RealSense user in that case found in https://github.com/IntelRealSense/librealsense/issues/900#issuecomment-394268551 that calling pc.calculate(depth) twice worked for them.  I'm very pleased that you were able to develop your own solution!

ilyak93, 2022-04-19:
@MartyG-RealSense, actually the same solution worked for me, forgot to mention that the whole working code looks like:

```
points = pc.calculate(depth);
auto color = frames.get_color_frame();
pc.map_to(color);
points = pc.calculate(depth);
```

So the second calculation gives the proper colorful pointcloud.

ilyak93, 2022-04-19:
@MartyG-RealSense, just a small following question, if I need the pixels of every point, do the coordinates and vertices which are return by those functions guaranteed to be aligned (in the same order) ?

```
const rs2::texture_coordinate* texcoords =  points.get_texture_coordinates();
const rs2::vertex* vertices =  points.get_vertices();
```

I've noticed also that when I save the pointcloud to ply, despite the fact that the size of the pointcloud in c++ showed to be full (i.e for an image of 720 x 1280, the size is 720 * 1280), the size of the saved ply pointcloud after reopening in matlab and python open3d lib both is much less, about 640k of the shown before 920k.

I've also noticed that that if I count all the point which texcoords arent (0,0) I get the exactly same amount of points which are actually saved (about 640k). 

So if I go over the points in such manner:

```
            for(int l = 0 ; l < sz; ++l) {
                if(texcoords[l].u != 0 || texcoords[l].v != 0){
                    cout << vertices[l].x << ' ' << vertices[l].y << ' ' << vertices[l].z << endl;
                    cout << texcoords[l].u << ' ' << texcoords[l].v << endl;
                }
            }
``` 

Is it guaranteed that each x,y,z corresponding to u,v and the if guarantees that I take only the actual points (that is concluded from the observation about the same amount of points in the saved ply) ?