realsenseai/librealsense #10413: cant connect to camera

NizarMhatli, 2022-04-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D455e } | 
| Firmware Version                   | (05.12.13.50) | 
| Operating System & Version |   {Win (10 | 
| SDK Version                            |  { 2.50.03785 }                          | 

new to Framos real sense camera 
installed and built the sdk and python wrapper following the instructions when i run the examples i still get this error:
the camera can be detected if i run the viewer from the program files folder 
 pipeline_profile = config.resolve(pipeline_wrapper)
RuntimeError: No device connected

MartyG-RealSense, 2022-04-19:
Hi @NizarMhatli  As the FRAMOS 'e' range of RealSense-compatible industrial cameras are not an Intel product, Intel does not provide technical support for them.  Please contact FRAMOS support using the **Contact Us** button on the page at the link below.

https://www.framos.com/en/resources/support

MartyG-RealSense, 2022-04-28:
Case closed due to D455e being a non-Intel camera product.

closed 2022-04-28 by MartyG-RealSense