realsenseai/librealsense #10414: Origin of world points using rs2_deproject_pixel_to_point()

harezhh, 2022-04-19:
Dear Community, 

I need to obtain the world position of an image pixe. Thats why I use rs2_deproject_pixel_to_point(). But I dont know where is the origin of the world point. Is it in the origin of the RGB-Camera or Depth-Transmitter?

Thanks.

MartyG-RealSense, 2022-04-19:
Hi @harezhh  The depth origin of RealSense 400 Series stereo depth cameras is the center line of the **left IR imager**.  Origin diagrams relating to different models in the 400 Series product range can be found on pages 86-88 of the current edition of the 400 Series data sheet document.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

On the L515 lidar depth camera, the depth origin is the center of the **IR Transmitter**.

[image]

harezhh, 2022-04-19:
Thank you very much for this quick response :) !

Ok, imagine I have the image pixel point (x,y) and I want to obtain the real world coordinate (X,Y,Z) relativ to the camera. Afterwards I want to obtain the error in Y-direction by measuring manually with a tape. As I understood, I have to position the beginning of the tape the center of the IR Transmitter ?

MartyG-RealSense, 2022-04-19:
On RealSense cameras, depth / distance is represented by the Z coordinate.   Depth is measured from the front glass of the camera instead of the camera lens behind the glass.  You can find the **ground truth** measurement value by adding **-4.5** (mm) to depth measurement values.

It may be harder to measure the real-world Y value with a tape measure, as the XY measurements would be in relation to how far the observed Y coordinate is from the depth origin point.  

As described by the RealSense SDK's **Projection** documentation (see the link below), the positive x-axis points to the right, the positive y-axis points down, and the positive z-axis points forward.

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#point-coordinates

So if a point is in the top left area of the image, it could have negative x and negative y values (because it is to the left of the imager center (minus X axis) and above the imager center (minus Y axis).  Therefore, a minus X and positive Y result would refer to a coordinate that is to the left and below the imager's center.

harezhh, 2022-04-19:
Thank you.

I did understand what you mean. But when I hold the tape on the L515. Should I hold it on the center of IR-Transmitter or center of RGB-Camera? Thats still not clear for me unfortunatelly.

MartyG-RealSense, 2022-04-19:
In regard to Z depth, you can extend the tape from the center of the IR Transmitter to the surface of the measured object.  The IR Transmitter would also be the origin point for the X and Y of deprojected XYZ coordinates.

When determining X and Y accuracy though, people usually use a best-guess estimate about the amount of error.  Sometimes the error is immediately obvious, such as 2 meters X or Y measurement for a coordinate that visually appears to be 3 cm away from the center of the image.

closed 2022-05-04 by MartyG-RealSense