realsenseai/librealsense #10415: Access left and right imagers in D405

DimitrisKld, 2022-04-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405  | 
| Firmware Version                   | 05.12.14.100 | 
| Operating System & Version |   Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    |  5.11.0-051100-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0                          | 
| Language                            |  python                       | 

### Issue Description

Hi, 
I want to get the left and right images from the left and right sensors of D405 using the librealsense SDK. I managed to obtain them using the following lines of code:
```
import pyrealsense2 as rs
import numpy as np

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.infrared, 1, 848, 480)
config.enable_stream(rs.stream.infrared, 2, 848, 480)
profile = pipeline.start(config)

left_frame = frames.get_infrared_frame(1)
right_frame = frames.get_infrared_frame(2)
left_img = np.asanyarray(left_frame.get_data())
right_img = np.asanyarray(right_frame.get_data())
```
However, I would like to ask you why I can access the images from the left and right sensors using the infrared stream, as the D405 does not have an infrared projector. 
Moreover, I would like to ask you if it possible to get color images from both left and right sensors, because I get grayscale images.

Thank you in advance.

MartyG-RealSense, 2022-04-19:
Hi @DimitrisKld  On 400 Series camera models that have a projector, it is a separate component from the left and right IR imagers and so infrared data can be obtained even when the projector is disabled.  

The role of the projector is to cast a semi-random pattern of dots onto surfaces in a scene so that the camera can use those dots as a texture source to analyze for depth information.  This is useful for obtaining depth from objects and surfaces that have low texture detail or no texture, such as doors, walls and desks.  

If the projector is disabled (or is not included on a particular RealSense camera model) then the amount of detail on the depth image can be reduced, making it more sparse.  400 Series cameras can though alternatively use ambient light in a well-lit scene to analyze surfaces for depth information instead of using the projector dots as a texture source.

The D405 model cannot see infrared dot patterns even if provided by another device, such as a different 400 Series model with a projector or an external infrared projector, as the sensors on the D405 have an **IR Cut** filter on them that makes infrared frequencies invisible to the camera.  It should be able to see a projected pattern from an external projector that casts a **visible pattern** though.

On RealSense camera models that support color images from the infrared sensor (D405, D415 and D455), color can only be obtained from the left infrared sensor in RGB8, BGR8, RGBA8 and BGRA8 formats.  Color is not available from both left and right infrared sensors.

michael-nc, 2022-04-19:
@DimitrisKld How are you able to access both infrared streams on D405? Using the intel realsense viewer, it shows that I only have access to a single IR stream (usb 3.2). 

[image]

MartyG-RealSense, 2022-04-19:
Hi @michael-nc  If you have a missing Infrared 2 stream on a USB 3.2 connection on Windows then it can indicate that there may be a problem with your two Depth and RGB camera drivers in the 'Cameras' section of the Windows Device Manager.  If it is that problem then a procedure at the link below to **completely** remove and re-install the drivers can correct it.

https://support.intelrealsense.com/hc/en-us/community/posts/4419989666323/comments/4431239847443

michael-nc, 2022-04-19:
> 

@MartyG-RealSense thanks, that worked!!!

MartyG-RealSense, 2022-04-19:
That's great to hear   :)

closed 2022-04-20 by DimitrisKld