realsenseai/librealsense #10416: Jetpack 5.0 support?

bblumberg, 2022-04-19:
----------------------------------------------------------------------------------------------------
Camera Model: D405
Operating Version: 20.04 (Jetpack 5.0)
Kernel Version: 5.10
Platform: Jetson Xavier AGX
SDK: latest
Language: Python
Segment: Robot

Question: Are there plans to release a compatible version of librealsense2 and pyrealsense2 for Jetpack 5.0, and in the meantime, any pointers on building a compatible version, and my chances of success. Thanks!

MartyG-RealSense, 2022-04-19:
Hi @bblumberg  At present the most recent supported Jetson configuration in librealsense version 2.50.0 is:

Ubuntu 18.04
JetPack 4.6
L4T 32.6.1

A Jetson NX user reported success with this configuration a month ago at the time of writing this.

https://support.intelrealsense.com/hc/en-us/community/posts/4744980703379/comments/4786889046035

They were subsequently able to successfully install pyrealsense2 with an instruction guide at https://github.com/IntelRealSense/librealsense/issues/7722

Whilst Jetpack 5.0 can be used with Ubuntu 20.04, librealsense does not currently have JetPack 5.0 support in SDK 2.50.0 and there is no information at this time regarding possible future support for it in the SDK.

bblumberg, 2022-04-19:
Thanks for the quick reply. Yes, I have had good luck as well with
realsense2.x using 4.6, and at the time, found the instructions on #7722
very helpful.

I have had good luck with realsense running on 20.04 on my intel linux
machine so I was hoping, at most, it would be a simple recompile of the
library for ARM, but I guess it is more involved than that. Is there any
point in trying, or is that a sure recipe for frustration?

bb

MartyG-RealSense, 2022-04-19:
There was a case in February 2022 at https://github.com/IntelRealSense/realsense-ros/issues/2213#issuecomment-1047696134 where somebody was running JetPack 4.6 on 18.04 in combination with 20.04 on a Docker setup.  In the course of researching that case, I found that Jetson developers had experimented with workarounds for getting 20.04 to work with 4.6 and shared the findings at https://github.com/IntelRealSense/realsense-ros/issues/2213#issuecomment-1050112918

I would personally recommend sticking with the configuration that is known to work, but you are of course free and welcome to perform your own experimentations like those workaround creators did.

bblumberg, 2022-04-19:
> There was a case in February 2022 at IntelRealSense/realsense-ros#2213
> (comment)
> where somebody was running JetPack 4.6 on 18.04 in combination with 20.04
> on a Docker setup. In the course of researching that case, I found that
> Jetson developers had experimented with workarounds for getting 20.04 to
> work with 4.6 and shared the findings at IntelRealSense/realsense-ros#2213
> (comment)
>
> I would personally recommend sticking with the configuration that is known
> to work, but you are of course free and welcome to perform your own
> experimentations like those workaround creators did.

Thanks. And, yes, good advice. There are a bunch of reasons for wanting to
be on 20.04, and now that this is officially (if still in developer preview
mode) supported, and the future of the Jetson platform, I am eager to move
over to it, if I can. Anyway, I will experiment and report back what I
learn. Thanks again.

bb

MartyG-RealSense, 2022-04-19:
You are very welcome.  I look forward to hearing about your test results.  Good luck!

koralm, 2022-04-23:
@[bblumberg](https://github.com/bblumberg) I'm currently testing 2.50.0 with jetpack 5.0 it looks that working without issues. even build form source with CUDA.

bblumberg, 2022-04-23:
Cool. Did you use the Linux instructions for the RSUB version? Thanks!!!!!

bb

koralm, 2022-04-23:
It looks that working both -DFORCE_RSUSB_BACKEND:=false and -DFORCE_RSUSB_BACKEND:=true.  But as is suggested in https://dev.intelrealsense.com/docs/build-configuration work better with -DFORCE_RSUSB_BACKEND:=false. I'm mainly testing on Jetson ANG NX with ROS noetic but it should work with Linux instructions for the RSUB.

MartyG-RealSense, 2022-04-29:
Hi @bblumberg  Do you require further assistance with this case, please?  Thanks!

bblumberg, 2022-04-29:
Close it out. Thanks.

MartyG-RealSense, 2022-04-29:
Thanks very much for the update!  As suggested, I will close the case. Thanks again!

closed 2022-04-29 by MartyG-RealSense

bblumberg, 2022-10-11:
Success here as well, so many thanks for leading the way. I followed the
linux instructions for building the force_rsub_backend option, and being
the python guy that I am, I also built the pyrealsense2 library. The viewer
works based on nosing around for 5 minutes, and while I need to play around
with the pyrealsense2 library it is promising that it built without
complaints and I was able to import it. The only build issues that I had
were: (a) I had to alias python to point to python3, and (b) I had to pull
in the libpython3.9-dev package because of a pybind11 dependency. But
all-in-all, unexpectedly easy.

This is a question for Marty, if you don't mind. I am confused about the
need for the kernel patches because in the default case where you apt
install the libraries, kernel patches are not applied are they?

Thanks again for everyone's help.

MartyG-RealSense, 2022-10-11:
Hi @bblumberg   With package installation, the kernel patch is bunded within the packages and so does not need to be applied separately.

zhjunqin, 2022-10-25:
> Success here as well, so many thanks for leading the way. I followed the linux instructions for building the force_rsub_backend option, and being the python guy that I am, I also built the [...]

Could you help share your full instructions to build librealsense and pyrealsense2  on Jetpack 5?

IamShubhamGupto, 2022-11-13:
> @[bblumberg](https://github.com/bblumberg) I'm currently testing 2.50.0 with jetpack 5.0 it looks that working without issues. even build form source with CUDA.

Could you provide some documentation how you have achieved this?
Thanks

AlbertoOddness, 2022-11-14:
> > @[bblumberg](https://github.com/bblumberg) I'm currently testing 2.50.0 with jetpack 5.0 it looks that working without issues. even build form source with CUDA.
> 
> Could you provide some documentation how you have achieved this? Thanks

I managed to build realsense using jetpack 5.0.2 following a few instructions from here https://github.com/IntelRealSense/librealsense/issues/7722#issue-736687248 and here https://github.com/IntelRealSense/librealsense/issues/6436#issuecomment-653942842 along with other things mentioned in this very issue

So essentially what I did was:

`sudo apt-get update && sudo apt-get -y upgrade
sudo apt-get install -y --no-install-recommends \
    python3 \
    python3-setuptools \
    python3-pip \
    python3-dev

sudo apt-get install -y git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev

sudo apt-get install -y libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev

sudo apt-get install libusb-1.0-0-dev

sudo apt-get install python3.9-dev

git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense/
mkdir build && cd build
cmake ../ -DFORCE_RSUSB_BACKEND=false -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
sudo make uninstall && make clean && make && sudo make install
`

The build was finished successfully, however I've yet to test it since I don't have a camera currently. If this works for you I'd like to know!

ryank-cobot, 2022-11-17:
Hello, we were able to build realsense using jetpack 5.0.2 as well using those instructions, however we are unable to successfully stream from multiple realsense cameras. Single Camera streaming works great however. Is there any guidance for how to enable multi camera streaming or any information on when this would be supported? Thanks so much for the help in advance!

MartyG-RealSense, 2022-11-17:
Hi @ryank-cobot   If you are using the RealSense SDK then you can test multiple cameras in the **RealSense Viewer** tool by left-clicking on the 'Add Source' button at the top of the Viewer's options side-panel and adding additional cameras one by one.

Multiple cameras on a Jetson board may be more stable if attached to a mains electricity powered USB 3 hub instead of attached directly to the Jetson or to a non-mains hub.

bblumberg, 2022-11-17:
My experience is that the D405 works great on the Orin. I haven’t had luck
with the D415/D435 (laggy, with missed frames, etc…) This is something I
didn’t see in earlier versions, e.g. 5.01.

bb

ryank-cobot, 2022-11-17:
Hi @MartyG-RealSense and @bblumberg Thanks so much for the responses. We seemed to have resolved our issues by switching to the D455. The D457 was not behaving well with USB.

MartyG-RealSense, 2022-11-17:
Hi @ryank-cobot   That's great to hear that you had success with D455.

The new D457 GMSL / FAKRA camera can be connected to USB for debugging and production line testing purposes but is designed for normal use with coax cabling.  https://github.com/IntelRealSense/librealsense/issues/11054 has cabling information.

ryank-cobot, 2022-11-17:
@MartyG-RealSense Awesome thanks so much for the quick responses. We really appreciate it.

ryanmaxwell96, 2022-11-29:
> sudo make uninstall && make clean && make && sudo make install

Awesome! Realsense camera D435i works on Ubuntu 20.04 on Jetson AGX using Jetpack 5.0.2. Just get one issue when I try to turn on the motion switch in Intel RealSense Viewer v2.52.1 - it says 

Backend in rs2_open_multiple(sensor:0xffff90017c70, profiles:0xffff9112e22count:2):
Failed to set frequency 1. device path:
/sys/devices/platform/310000.xhci/usb1/1-2/1-2:1.5/0003:8086:0B3A.0003/
Last Error: Permission denied

MartyG-RealSense, 2022-11-29:
Hi @ryanmaxwell96  IMU (motion module) problems have been previously reported with JetPack 5, as the most recent JetPack version support update was for **4.6** in librealsense SDK 2.50.0.

zachrizzo, 2022-12-02:
This worked for me as well and I also getting an IMU error, seems only like a permission thing though, ill dive in and see what i can do. otherwise i am going to buy an IMU from amazon

MartyG-RealSense, 2022-12-02:
Thanks very much @zachrizzo for your feedback about JetPack 5!

zachrizzo, 2022-12-02:
so after some messing around i was able to get the IMU working if i ran sudo in front of the command to open the Intel real sense Viewer
` sudo realsense-viewer`

I also deep dive into the files that were lacking permissions to see if i could change these permanently. However,  once you change the permissions and unplug the camera it default back to the original permissions. so far only thing that works is sudo.still haven't tested this with Ros 2 however.

zachrizzo, 2022-12-03:
also note you will get a missing config file error for the IMU, this is what worked for me
`# back to librealsense folder
sudo ./scripts/setup_udev_rules.sh `

does anyone know how to import this now from as a python module ?

bblumberg, 2022-12-03:
Try:

import pyrealsense2.pyrealsense2 as rs

bb

nakai-omer, 2023-02-12:
> ```
> sudo make uninstall && make clean && make && sudo make install
> ```

We has the following error/warning:
```
-- Checking for module 'libudev'
--   No package 'libudev' found
```
No also had to run:
```
sudo apt-get install -y libudev-dev
```

MartyG-RealSense, 2023-06-07:
Hi @bblumberg  The new 2.54.1 version of librealsense with JetPack 5.0.2 support is now available.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

Release notes:
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2541