realsenseai/librealsense #10417: Kinfu example fails with SIG_ABORT

Bangybug, 2022-04-19:
Hi Everyone,

System: D435, MacBook Pro (15-inch, 2018) with dual graphics Radeon Pro 555X 4 GB and Intel UHD Graphics 630 1536 MB.

Problem: When I run rs-kinfu, it works for some time, I can see few points in the cloud and can rotate it, but then it fails on 8-th or 9-th frame with SIGABRT from kf->update.

Has anyone tried running this sample recently? I cloned opencv and opencv_contrib from the master branch, which is the version above 4.5 I believe.

Here is how I build:

1) OpenCV

export MY_OPENCV_BUILD=~/Documents/workspaces/realsense/opencv_build
export MY_OPENCV_CONTRIB=~/Documents/workspaces/realsense/opencv_contrib
export MY_OPENCV=~/Documents/workspaces/realsense/opencv

mkdir $MY_OPENCV_BUILD
cd $MY_OPENCV_BUILD
cmake -DOPENCV_ENABLE_NONFREE=ON -DOPENCV_EXTRA_MODULES_PATH=$MY_OPENCV_CONTRIB/modules $MY_OPENCV
make -j5

cp -r $MY_OPENCV_BUILD/3rdparty/lib/*.* $MY_OPENCV_BUILD/lib

The last command copies libraries from the 3rdparty/lib directory to the lib directory. If I don't do that, the Realsense won't build because it looks for these libraries in /lib folder, not where they got built.  

2) Realsense

I got this info from two pages: https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk
and https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu

brew install libusb pkg-config
brew install homebrew/core/glfw3
brew install cmake

export MY_REALSENSE=~/Documents/workspaces/realsense
mkdir $MY_REALSENSE/build && cd $MY_REALSENSE/build
sudo xcode-select --reset
export OpenCV_DIR=$MY_OPENCV_BUILD
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_CV_EXAMPLES=true -DBUILD_CV_KINFU_EXAMPLE=true -DBUILD_GRAPHICAL_EXAMPLES=true  -G Xcode

MartyG-RealSense, 2022-04-19:
Hi @Bangybug  In the past, if there have been problems with running rs-kinfu then it has been because it was having problems with a line that tries to load the **High Density** camera configuration preset.  Editing this line out of the code usually fixes the problem, as described at https://github.com/IntelRealSense/librealsense/issues/8931#issuecomment-830063845 

Given that you are the first RealSense user I am aware of who has tried to run rs-kinfu on Mac, this preset may not necessarily be the cause of your particular problem though.

Bangybug, 2022-04-19:
@MartyG-RealSense thanks for the reply and well spotted!  I commented out that line  in rs-kinfu.cpp and it solved my problem, at least it works but very slowly and kinfu may reset sometimes. 

So this is something related to the capabilities of OpenCL which is used by OpenCV.  I also found a more relevant issue https://github.com/opencv/opencv_contrib/issues/2029 - they suggest some workarounds, but in general not much of help.

Just made a debug OpenCV build. It seems that the OpenCV is using Intel UHD graphics, but not the Radeon GPU, as follows from these logs

```
Metal API Validation Enabled
flock failed to lock maps file: errno = 35
flock failed to lock maps file: errno = 35
[ INFO:0@4.966] global /realsense/opencv/modules/core/src/ocl.cpp (1186) haveOpenCL Initialize OpenCL runtime...
[ INFO:0@4.966] global /realsense/opencv/modules/core/src/ocl.cpp (1192) haveOpenCL OpenCL: found 1 platforms
[ INFO:0@4.966] global /realsense/opencv/modules/core/src/ocl.cpp (984) getInitializedExecutionContext OpenCL: initializing thread execution context
[ INFO:0@4.966] global /realsense/opencv/modules/core/src/ocl.cpp (994) getInitializedExecutionContext OpenCL: creating new execution context...
2022-04-19 23:06:41.801034+0500 rs-kinfu[88930:6330712] [CL_INVALID_OPERATION] : OpenCL Error : Failed to retrieve device information! Invalid enumerated value!

[ INFO:0@4.970] global /realsense/opencv/modules/core/src/ocl.cpp (1012) getInitializedExecutionContext OpenCL: device=Intel(R) UHD Graphics 630
[ INFO:0@4.970] global /realsense/opencv/modules/core/src/ocl.cpp (5370) __init_buffer_pools OpenCL: Initializing buffer pool for context@0 with max capacity: poolSize=0 poolSizeHostPtr=0
[ INFO:0@4.978] global /realsense/opencv/modules/core/src/parallel/registry_parallel.impl.hpp (96) ParallelBackendRegistry core(parallel): Enabled backends(3, sorted by priority): ONETBB(1000); TBB(990); OPENMP(980)
exporting to041922 230654.ply
reset
exporting to041922 230728.ply
reset
exporting to041922 230818.ply
Program ended with exit code: 0
```

I think this issue has to be closed, as it's more related to OpenCV and OpenCL.

closed 2022-04-19 by Bangybug

MartyG-RealSense, 2022-04-20:
Thanks very much @Bangybug for the detailed feedback!  It's excellent to hear that the edit to the preset line worked for you  :)