realsenseai/librealsense #10418: cannot calibrated IMU on D435i

w407022008, 2022-04-19:
Hello,
following the document [Intel_RealSense_Depth_D435i_IMU_Calibration](https://www.intelrealsense.com/wp-content/uploads/2019/07/Intel_RealSense_Depth_D435i_IMU_Calibration.pdf ), I cannot calibrate the imu because it get always stuck there shown below. I have been waiting for more than ten minutes, but it still can't start the next step even though it shows that all three directions are aligned. 
OS: Ubuntu 20.04
Python: python3
librealsense: 2.44.0 (I'm working on this version)

```
waiting for realsense device...
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  046222070603
  Product Line:  D400
  Firmware version:  05.12.11.00
Start interactive mode:
FOUND ACCEL with fps=63
FOUND GYRO with fps=200
-------------------------
*** Press ESC to Quit ***
-------------------------

Align to direction:  [ 0. -1.  0.]   Mounting screw pointing down, device facing out
   Status.rotate:           [ 0.     -0.0001 -0.0148]:                 [ True  True  True]
Error: User Abort.
```

MartyG-RealSense, 2022-04-19:
Hi @w407022008  When using the IMU calibration tool, the user should turn the camera until the **Status.rotate** numbers closely match the **Align to direction** set of target numbers above it.  Once you have matched the live-updating Status.rotate values as closely as possible with the fixed Align to direction values then you let go of the camera.

If the match is decided by the program to be close enough to be successful then a series of full-stop dots starts printing on screen.  Once this accumulating line of dots reaches a certain length then the program concludes that calibration phase and asks you to then turn the camera to a different set of target values.  

Once this has been performed for six different camera orientations, the calibration process concludes and you are asked whether you wish to permanently write the calibration to the camera hardware to preserve it or start a new calibration.

In your calibration above, it looks as though there is not a close match to the requested middle value of -1, as your middle angle value is **-0.0001**.  This may be why the calibration is not progressing to the next stage.

w407022008, 2022-04-19:
Very strange. I don't get a value close to (0,-1,0) no matter how I rotate the camera.

w407022008, 2022-04-19:
If Mounting screw pointing up, device facing out, the terminal output:
   Status.rotate:           [ 0.0048 -1.9992 -0.04  ]:                 [ True False  True]

w407022008, 2022-04-19:
Do you have a video tutorial? I have rotated the camera many times and only this one case shows three true. So I'm sure I've followed the requirements (Mounting screw pointing down, device facing out). But it does not execute the next step, even though I updated the version of librealsense upto 2.50.0 and firmware to 05.13.00.50.

MartyG-RealSense, 2022-04-19:
The camera has to be in a specific orientation on each of the six phases of calibration.  These orientations are clearly illustrated in the PDF version of the user guide for the IMU calibration tool at the link below, which is the best guide for using this tool in my opinion.

https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

The illustration of the positions begins at the bottom of page 16 of the PDF guide.

w407022008, 2022-04-19:
Hello,
I got the reason! 
When I checked the source code [here](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/tools/rs-imu-calibration/rs-imu-calibration.py#L173), I found that the algorithm sets the maximum max_norm to [0.5,0.5,0.5]. However, inevitably there are always some sensors that have a large deviation, for example, the original output of my sensor is [0,8.9,0] compared to the target [0.9.8.0] which is 0.9 worse than the set value, so the algorithm always thinks that it is not placed correctly. In fact this condition (is_in_norm) is unnecessary and can be completely removed or the user can be asked to change the set value if necessary.
Pls improve it.

closed 2022-04-19 by w407022008

MartyG-RealSense, 2022-04-19:
It's great to hear that you achieved a solution through your edit!  Thanks so much for the update and for sharing details of your fix.

In regard to improving the tool's script, the best course of action would be to create and submit a **Pull Request (PR)** of your code change so that the RealSense development team can review and consider the change for merging into the SDK.

https://github.com/IntelRealSense/librealsense/pulls