realsenseai/librealsense #10419: Getting for all the (x,y,z) points in a point cloud their image_x, image_y coordinates (pixel location)

ilyak93, 2022-04-19:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D465                                                       | 
| Operating System & Version | Win 10                                                    | 
| Platform                                 | PC                                                           |
| SDK Version                           | 2.48.0.3381                                             | 
| Language                               |  C++                                                       | 
 

I need the pixels of every point in a point cloud, the first thing I though about is using:

```
const rs2::texture_coordinate* texcoords =  points.get_texture_coordinates();
const rs2::vertex* vertices =  points.get_vertices();
```
But I noticed that u and v aren't integers, so I don't understand what are they exactly. I thought those are image_x and image_y coordinates (pixel location), but obviously it isn't. 

Other approach is to calculate the pointcloud as I did in issue #10412:
```
points = pc.calculate(depth);
auto color = frames.get_color_frame();
pc.map_to(color);
points = pc.calculate(depth);
```

 And then project points to pixels:

```
            float flpixel[2];
            int pixel[2];

            for(int l = 0 ; l < sz; ++l) {
                rs2_project_point_to_pixel(flpixel, &depth_intrinsics, vertices[l]);
                pixel[0] = int(flpixel[0]);
                pixel[1] = int(flpixel[1]);

                if(!std::isnan(flpixel[0]) && !std::isnan(flpixel[0])){
                    cur_xyz_matrix[pixel[0]*W+pixel[1]+0] = vertices[l].x;
                    cur_xyz_matrix[pixel[0]*W+pixel[1]+1] = vertices[l].y;
                    cur_xyz_matrix[pixel[0]*W+pixel[1]+2] = vertices[l].z;
                }
            }
```

Here I have few questions:

1. Should I use the depth intrisics or the rgb intrinsics ? 
2. If I do it the way presented above I get i.e 620k points in the pointcloud and if I used the depth intrinsics, do I understand properly that the pixel locations that I got by projection are the x and y of the depth image ?
3. Following 2, if the x and y are of the depth image and I need x and y which corresponding to my rgb image, how should project the points ?

My understanding and suggestions about Q.3:

1. I can align depth to color first and only after that `calculate` the point cloud, which works at first look, but I get 200k points less. Which isn't surprising because during the alignment some of the depth points are lost because the resolution of depth initially is zoomed-like, so it is zoomed out during the alignment, so it is aligned horizontally and vertically but also down-sampled as much as I understand.

If I calculate the projection using the rgb_intrinsics I get the same amount of points which I get with depth_intrinsics, but the numbers sometimes are float which isn't the case using the depth intrinsics (I get only integer values, which makes sense because it's pixel location, and they are different comparing color_intrinsics projected and depth_intrinsics projected), is this a sign the depth intrinsics are the only correct to use ? Or maybe if I round the float numbers I get the correct x,y pixel location for the point x,y,z projected from the point cloud ? 

I wonder if I can somehow not lose those 200k points because of the alignment. I also wonder when I map color to the depth using rs2::pointcloud and I have a colorfull pointcloud, what are those 200k additional points I have comparing to the aligned frames pointcloud ?

It sounds that the colorful pointcloud contain more information, but I can only get the x,y pixel location of the depth which is not aligned to the color. 

Any suggestions ?

I'll probably stay with the solution of alignment, so I asking if I can somehow get the pixel locations of the points in the pointcloud computed by the code above:

```
points = pc.calculate(depth);
auto color = frames.get_color_frame();
pc.map_to(color);
points = pc.calculate(depth);

```

Example of aligned and non-aligned depth and color images:

[image]
[image]

aligned:
[image]

same images but aligned(down-sample effect):

ilyak93, 2022-04-19:
I've tried the opposite color_to_depth alignment and after it the pointcloud creation, so this mostly solves the problem.

[image]

They

ilyak93, 2022-04-19:
I've tried to project points to pixels with `rs2_project_point_to_pixel` and then to reproject them back to points with `rs2_deproject_pixel_to_point` and I didn't received the same points. 

I did alignment first which worked well and used depth_intrinsics in both projection and reprojection.

The code:
```
            fs = align.process(fs);
            rs2::frame depth_frame = fs.get_depth_frame();
            rs2::frame color_frame = fs.get_color_frame();

            rs2::pointcloud pc;
            rs2::points points;
            points = pc.calculate(depth_frame);
            pc.map_to(color_frame);
            points = pc.calculate(depth_frame);

            const rs2::vertex* vertices =  points.get_vertices();

            size_t sz = points.size();

            for(int l = 0 ; l < H*W; l+=3) {
                cur_xyz_matrix[l+0] = std::numeric_limits<double>::quiet_NaN();
                cur_xyz_matrix[l+1] = std::numeric_limits<double>::quiet_NaN();
                cur_xyz_matrix[l+2] = std::numeric_limits<double>::quiet_NaN();
            }

            float flpixel[2];
            int pixel[2];

            for(int l = 0 ; l < sz; ++l) {
                rs2_project_point_to_pixel(flpixel, &depth_intrinsics, vertices[l]);

                pixel[0] = int(flpixel[0]);
                pixel[1] = int(flpixel[1]);

                if(!std::isnan(flpixel[0]) && !std::isnan(flpixel[1])){

                    std::pair <int,int> cur_p(pixel[0],pixel[1]);

                    cur_xyz_matrix[pixel[0]*W+pixel[1]+0] = vertices[l].x;
                    cur_xyz_matrix[pixel[0]*W+pixel[1]+1] = vertices[l].y;
                    cur_xyz_matrix[pixel[0]*W+pixel[1]+2] = vertices[l].z;

                    float reprojected_point[3];

                    unsigned short tmp = depth_image.at<unsigned short>(pixel[0], pixel[1]);
                    float dep = (float) tmp;
                    rs2_deproject_pixel_to_point(dereprojected_point, &depth_intrinsics, flpixel, dep);

                  //the result is that vertices[l] != reprojected_point
                  //pixel[0] in this exmaple is 100 and pixel[1] is 0
                }
            }
```

ilyak93, 2022-04-19:
Ok, first thing I have found that for my image of 720x1280 the function `rs2_project_point_to_pixel(flpixel, &depth_intrinsics, point);`

returned into flpixel coordiantes that were (934, 0) so I suppose the coordinates are returned as (y,x) and not (x,y). I'll continue to research it.

MartyG-RealSense, 2022-04-20:
Hi @ilyak93  There is a lot of information to digest in this case.

If you have an aligned image and you use rs2_deproject_pixel_to_point to get 3D points then there is a risk of inaccuracy being introduced during the deprojection due to alignment, as commented on by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/4315#issuecomment-518385243 and https://github.com/IntelRealSense/librealsense/issues/2523#issuecomment-430246582

Once inaccuracy is introduced during the initial deprojection from pixel to point then those inaccuracies will be factored into the values when projecting back from point to pixel.

Using get_vertices can therefore provide better accuracy than rs2_deproject_pixel_to_point

MartyG-RealSense, 2022-05-04:
Case closed due to no further comments received.

closed 2022-05-04 by MartyG-RealSense

Amayess, 2022-07-01:
Hi @ilyak93, 

I am trying to understand and reproduce your example since I am looking for the same result as you : getting the pixel location of each points of the PCL.

Actually, I have a some understanding issues like : 

- What's the purpose of the cur_xyz_matrix and how did you declared it ? 
- Did you use the OpenCV library for this example ?
- Were you able to get the location of each pixels in each points of the PCL ? 

Thank you.

ilyak93, 2022-07-01:
@Amayess, I was able eventually, but not in the fashion I tried in this post. The vector of points eventually you have is the vector of sequentially ordered points of those pixels for which the depth isn't 0 in the corresponding depth frame.
So you must save it also separately. Then the mask `depth !=0` gives you for each point the actual pixel location.

Amayess, 2022-07-04:
Hi @ilyak93, Thanks for your answer but it's a little bit abstract for me, is there any snippet of the code so i can more visualize it and understand it ? 

Thank you.

Amayess, 2022-07-04:
So far now, this is what i did in order to get the pixel location corresponding to each point of the PCL.

Is my approach correct ? @ilyak93 

[image]

MartyG-RealSense, 2022-07-11:
Hi @Amayess  Do you require further assistance with your problem in this case, please?  Thanks!

Amayess, 2022-07-11:
Hi @MartyG-RealSense, problem solved few days ago, I forgot to update it.

Thank you !

MartyG-RealSense, 2022-07-11:
That's great to hear, @Amayess  - thanks very much for the update!